wled/WLED #1319: Custom build for ESP32; GPIO0 on ESP32 is GPIO17 (QuinLED Dig-Uno)

palengh, 2020-11-06:
When using ESP32, and in order to use the GPIO0 pin on the Dig-Uno we need to change it to be pointing to GPIO17, not GPIO16, which is working on the ESP8266. I would use this particularly to turn light on/off.

Is it possible to make a new build for this, or even better: being able to change this on the fly in the software like other settings.
The same way would be great to control all accessible GPIOs for connecting/defining different sensors like motion, temperature and others.

huggy-d1, 2020-11-06:
@Aircoookie has posted many times the requested enhancement will be a standard feature around release 0.12.  Since he's a student actively engaged in academic pursuits, and there's only so many hours in the day, it may be 2021 before that happens.  In reality, it may be longer than anyone would like, _especially_ @Aircoookie.  

If you are a skilled ESP8266/ESP32 developer, please engage the WLED developer community, get involved and lend a hand.  

Nobody will mind having additional perspectives, strengths, and developer availability to add to the current mix.

You can use VS Code / PlatformIO to download the WLED source, make a couple adjustments to the pins, and build your own custom firmware.  If it's specifically geared towards enhancing QuinLED Dig-Uno's, check out @Quindor's discord server and see if someone there can build it for you.

peacepenguin, 2020-11-07:
I would suggest getting yourself familiar with platformio building and uploading from the source code. The change your after is probly really easy. I don't fully understand what your asking, but here's a snip for a specific "wemos" board that overrides some pins simply by adjusting the "build_flags".

```
[env:wemos_shield_esp32]
board = esp32dev
platform = espressif32@1.12.4
upload_port = /dev/cu.SLAB_USBtoUART
monitor_port = /dev/cu.SLAB_USBtoUART
upload_speed = 460800
build_flags = ${common.build_flags_esp32} -D LEDPIN=16 -D RLYPIN=19 -D BTNPIN=17
lib_ignore =
  ESPAsyncTCP
  ESPAsyncUDP
```

Also you can checkout my fork for multipin output where i'm overriding LOTS of pins for outputs to get some ideas on how to change the source code to meet your needs:  https://github.com/peacepenguin/WLED

Here's a quick start guide on how to use platformio to compile and upload from the source code to get you started quickly in digging into the code. :
https://github.com/Aircoookie/WLED/issues/104#issuecomment-699557741

On the step for 'git clone' use the main URL instead of my fork'd copy, unless you want multi pin support too:

`
git clone https://github.com/Aircoookie/WLED
`

Personally I find using platform io to build and upload the code to be the easiest way to install most ESP32 projects, including WLED. platformio handles all the nitty gritty details for you. You just have to learn platformio instead, which will be beneficial in the long run.

Good luck!

closed 2020-12-02 by Aircoookie