wled/WLED #1320: Fireworks 1D not working on 3rd of 3 segments

jezmck, 2020-11-06:
_Thank you so much for WLED and all your hard work, I post this only in an effort to help WLED continue to be the best._

Best described with a screenshot to start:
[image]

The third segment (LEDs 330..445, i.e. 115 LEDs) is all lit in the primary colour, and this accurately reflects the actual LEDs.

I can consistently reproduce this by installing the firmware, setting up the segments and selecting the **Fireworks 1D** effect.

If I reduce the size of the 3rd segment to 31 LEDs, the effect will start working, but obviously not using the whole string.

 - Board: TOOGOO ESP8266 D1 Mini
 - Version: 0.10.2
 - Format: WLED_0.10.2_ESP8266.bin

Aircoookie, 2020-11-06:
Thank you for the report!

Please see https://github.com/Aircoookie/WLED/issues/1236 :)

jezmck, 2020-11-08:
Thank you.
I did search, but perhaps only open issues, sorry!

closed 2020-11-08 by jezmck