wled/WLED #1321: Sync not reliably working

bjprice4par, 2020-11-06:
I have read through the forums and haven't been able to get the help I need.  First, thanks to @Aircoookie for this great piece of software.  My family is really enjoying it.  

My problem is with sync.  I have two 8622's running WLED (latest release).  One for my upper roof line (567 led's) and one for my lower roof line (570) led's.  Sync works some of the time.  In the is instances that it doesn't work, if I change the intensity or speed of the effect, it gets them back in sync.  The  Node MCU's are both right next to each other in a telecom box with the power supply, and have strong wifi signal.  I have checked the box to send notification twice, and it seemed to help, but still only about 60% of the time.  For example if I set the preset cycle to cycle through 4 presets about 6 out of 10 time the the two roof lines stay in sync.  Any direction on what I can do to have the sync more reliable?

rfordhamjr, 2020-11-24:
Hay Luck with this? Im having the same problem.

montie-001, 2020-12-09:
Same problem as well.  Its totally random if it works or not.  Udp doesn't seem to be reliable.

closed 2021-06-16 by stale[bot]