wled/WLED #1322: Coding style guidelines

catlee, 2020-11-08:
**Is your feature request related to a problem? Please describe.**
A consistent coding style can help new contributors understand code, and improves readability overall.

**Describe the solution you'd like**
WLED should adopt a coding style, and have it enforced by checks in CI. Most code editors can automatically format code according to a particular style.

I know that `clang-format` is fairly popular for formatting C++ code, but there are a few others as well.

huggy-d1, 2020-11-09:
I agree there should be a single consistent coding style, and it should be adopted.  However, enforcing it - my opinion - might not be the best solution (for WLED).

If someone (like myself) makes a useful PR, but does not properly implement the coding style - not for lack of trying, but from lack of know how - only to be stopped dead while trying to submit the PR.

My lack of understanding why subTract (might be) (is) a bad coding style reject reason should not be a reason to reject a PR.  Otherwise, I (again, just as an example) might run out of time and energy while trying to understand what's syntax-wise wrong with the PR only to find out it's not a syntax problem, but a style convention rejection.

Perhaps novices and n00bs in the coding world should be punished for bad coding styles?

Or, is there more to what you are suggesting where this can transparently be handled during merges?

catlee, 2020-11-10:
Perhaps if we limit scope to code format, and not all aspects of coding style, it makes the problem simpler?

I think an auto-formatter can go a long way to making this easier. I know VSCode supports automatic formatting via plugins like https://marketplace.visualstudio.com/items?itemName=xaver.clang-format.

The Arduino IDE uses this I think? https://github.com/arduino/Arduino/wiki/Arduino-IDE-Auto-Format-configuration

In addition, you can have git precommit hooks set up that will autoformat the code for you before submitting your PR.

CI could check the format of PRs, and if they fail, emit a message about how to fix it.