wled/WLED #1323: Intermittent LED operation

bluedke70, 2020-11-09:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

bluedke70, 2020-11-09:
I am running 5 volt LED strips off a 60 amp 5v power supply.  WS2812B strips 30 leds per meter and I am running them in parallel strands, one is 500 LEDs the other is 600 leds   last year they worked perfectly, this year when I went to start them up they did not.  So I rewired the 2866 node mcu and logic level shifter and it worked again, but only about half of the way down the strand. I changed the number of leds in the configuration but this didn’t help.  I went out later and it was working almost all the way down the run now.  But then I tried it again and it only worked to a different spot in the string.  Downstream of that point there are random leds lit and are not being controlled by the node mcu.  But that spot keeps changing. I took off the automatic dimming feature and that helped it work, but still not all the way down the run.  This feels like a power supply issue to me since I did nothing to that side EXCEPT turn power on and off.  But the power supply reads 5 volt output, but the program will not estimate the output amperage.  Any input would be appreciated.  Thanks again.

sansillusion, 2020-11-09:
To me it sounds like welding degredation on the strip. I might be wrong but
check to see if the string fails after a welded joint or a random led
between welds. It could give an idea.

bluedke70, 2020-11-09:
I expected the same thing but where it initially stopped there was no splice.  As a matter of fact none of the points where the lights stopped working were at splice points.  Thats why I was down to the power supply.  To me an intermittent power supply seemed to fit, especially after I turned off the automatic light dimmer function and they started to work again, but not all the way down the string.  I was wondering if I fried it somehow last year

huggy-d1, 2020-11-09:
How many power injection points do you have on each of the 2 strips?  If you disconnect either strip, does the remaining strip behave exactly the same (as originally posted)?

bluedke70, 2020-11-09:
Each side has at least 4 injection points. I have disconnected one splice point and now the downstream doesn’t work so I think that is a different issue. Before I disconnected it it did the same thing as the other side

bluedke70, 2020-11-24:
Thanks for the help.  I found a couple bad LED strings. Replaced these and they work

closed 2020-11-24 by bluedke70