wled/WLED #1324: MQTT command to reboot WLED

psbest, 2020-11-09:
I have noticed that the HTTP interface is not always available on bad wifi signals (long distance garden), but the MQTT interface always works.  After a reboot (manual reset button) the HTTP mostly works again. 
Is it possible to reset the WLED with a API/MQTT command, so i dont have to go into the garden :-) I have looked at the examples and wiki but was not listed.

Aircoookie, 2020-11-10:
Hi!
Yes, sending `RB` to the API topic should reboot your device (sorry for the connectivity issues in the first place, I suspect it is a problem ESPAsyncTCP might have with intermittent wifi connections)

Added that to the docs!

psbest, 2020-11-10:
YES it work, thank you, this is really great. Many thanks for the wled software, having a lot of fun with it!

closed 2021-06-06 by stale[bot]