wled/WLED #1326: Has the JSON string a max length?

StefanRu1, 2020-11-09:
Hi,

i have a problem with the json string.
I use Enigmalight on a Dreambox together with WLED to make LEDs light up in the color of the content on my TV screen.
For this a wrote a small python script and send a json request with the array of the colors. 
Like described in the wiki: {"seg":{"i":[[255,0,0], [0,255,0], [0,0,255]]}}
But i do this for all LEDs in my string. So it gets a pretty long json string.

This works very well.
I have a lamp with 100 lights and it works very good.

Now i wanted to add my floor lamp too.
This lamp has 154 LEDs.
And it did not work. I tried a lot and recognized that 121 LEDs did work but not 154.
This seems to be related to the length of the color array.
I am not very good at python but tried my best to get a response and reason.
By doing getdata() on the response object i get for a lamp that works:
Response: {"success":true}

Doing this when i have my lamp with 154 LEDs i get:
Response: {"error":10}

Can you tell me what error:10 means? 
Do you have an idea how long a jason push can be?
I am pushing with:

huggy-d1, 2020-11-09:
Try splitting the json data into PUT requests less than ~125 LEDs per PUT.  Might be the packet size causing problems.  Might be the amount of time available to process the packet.  Are you using ESP32?  If so, then packet size / speed should not matter as much.  If you are using ESP8266 or similar, then keeping things small and simple.

For example, when loading a bunch of segments a pixel at a time, turn off the LEDs so WLED can spend more time on the segment work.

You might consider xLights or a smaller variant that implements E1.31 protocol and configure everything precisely in that software, and then have it control WLED.

StefanRu1, 2020-11-10:
Yes i am using  ESP8266.
I was very impressed how good sending a json request is performing.
So as long as i stay under 125 LEDs it is very fast and no problem to process the requests.
I will implement a logic to split the request in to more parts with 100 LEDs per part.

I would like to use another protocol like E1.31 but would have to implement it in python and do not know where to start.
Regretful i can only use EnigmaLight which does not support this protocols native.

It supports output /dev/ttyUSB0 with type momo, which works well für cable connected WLED.
But my other lamps are in the WLAN and there the only way is to call a python script out of Enigmalight.
output          python /usr/wled_all_led.py 
type            popen
This sends all the light information with popen to python and i can build in the python script the json requests.

As said i am very happy with the performance and will try to get the json requests into parts.

Thanks alot, i will try and update her and close if it is working fine.

Best regards,
Stefan

Aircoookie, 2020-11-10:
Hi Stefan, indeed there is a limit both on the incoming string lenght (I believe it is 16kB, though I'm not exactly sure), and on the size of the JsonDocument created by the ArduinoJson library, which is 8192 bytes. This limits not the input string size, but the size after the input is de-serialized/parsed by WLED. You can enter your JSON [here](https://arduinojson.org/v6/assistant/) to see how much space it would use.

For your use-case I would indeed recommend one of the UDP based protocols WLED supports. E1.31 is rather on the complicated side, DRGB is the simplest one and should be easy to make a python script for. You just need a udp socket to port `21324` and set the first two bytes of the packet to `1`. After that, just the R, G, B of each LED - https://github.com/Aircoookie/WLED/wiki/UDP-Realtime-Control#udp-realtime

The `i` feature of the JSON API was meant more for little status indicator LEDs and such that can be set remotely - not originally for updating the entire string a few times per second - but if you manage to get that working, that's great too 👍 

Kind regards
Christian

StefanRu1, 2020-11-10:
Hi Christian,

thanks for your help. That's great information for me.
Yes i know using json is more a hack than a good implementation.

So i will go for UDP with DRGB. I like the way the UI behaves and that it recognized the connection and you can override the connection in the UI. So indeed this is what i am searching for and want to go with.
I will try to figure out how to do the UDP connection in python.

Thanks again!

Aircoookie, 2020-11-10:
Awesome!

Just a little correction, for DRGB you actually need the first byte to be `2` and the second `1` :)

StefanRu1, 2020-11-10:
Hi Christian,

this was easier than i thought. It is working great!
I drive now 2 lamps with over 350 LEDs.
I use now DRGB, that was the easiest way.
Perfect thanks for pushing me in this direction.

I will publish the script on a forum for EnigmaLight. If wanted i can also write the link here.
The script in this state makes EnigmaLight working over WLAN with WLED, which was not possible till now.

Thanks again!

StefanRu1, 2020-11-11:
Here the link to the post with the python script to drive WLED via DRGB UDP for EnigmaLight.
https://board.newnigma2.to/wbb4/index.php/Thread/32156-EnigmaLight-pclin-edition/?postID=330652#post330652
You will find also the EnigmaLight plugin for the Dremabox and all other information provided by pclin in this thread.

EnigmaLight is a plugin for Enigma2 (Linux) TV receivers to drive a ambilight.
The content of the screen is scanned and the LEDs can be driven acordingly.

Thanks and best regards,
Stefan

closed 2020-11-11 by StefanRu1