wled/WLED #1327: NTP time stopped working , no way to setup time

jsonoine1111, 2020-11-09:
Initially NTP server was working fine (after first first setup).
After electricity outage 2 days later NTP server sync is not working anymore: tried different servers, direct ip/name, different timezones, removed EPS8266 from QuinledUno board, did factory reset, reflashed same WLED_0.10.2_ESP8266 firmware etc.
Currently time displayed is 1970-1-1 for CET and 2106 for US-AZ. 
Please add option to setup time manually. I do not need precision to the second, I just want to schedule my lights.
Thanks.

I found simiar post here:
https://wled.discourse.group/t/ntp-time/819

huggy-d1, 2020-11-09:
Does your home wifi router provide NTP services?  Try that.  
Does any computer on your home network stay online most of the time?  If so, enable NTP services on that (google for your specific operating system) and point to that device.
Do you have a raspberry PI you can dedicate to WLED NTP services (among other things)?  If so, enable NTP services and point to that.

jsonoine1111, 2020-11-09:
Hi! Thanks for quick response. 
_Does your home wifi router provide NTP services? Try that._ This was already part of testing. I wrote here after tried all options available.

jsonoine1111, 2020-11-10:
NTP is syncing fine today again. Nothing was changed. Just enabled NTP server in config and received proper current local time.

Aircoookie, 2020-11-10:
Hi, 
that is very strange, I would have suspected DNS issues or something wrong with the timeserver, but as you tried several that could not be the issue (if you rebooted WLED after saving the new server).
Maybe something is/was wrong with the ESP modules internet access (firewall?)
But i'm glad to hear that it is working again for now. Keep an eye out for it!

Manual time sync is not as useful as NTP as it has to be done after every reboot. But this is still a good idea, I will make the UI client (phones and PCs are pretty much guaranteed to report the correct time) transmit the timestamp to the WLED device if a control command is send :)

Aircoookie, 2021-08-06:
Manual time sync when a command is sent via the web UI is now supported. Closing this for now.

closed 2021-08-06 by Aircoookie