wled/WLED #1329: Option for one Alexa device per Segment

Aircoookie, 2020-11-10:
**Is your feature request related to a problem? Please describe.**
For installations that consists of multiple parts, segments are very useful to control those parts individually.
Voice control of individual would thus also be handy, especially with the upcoming 0.12.0 multipin support.

**Describe the solution you'd like**
One emulated Alexa device per (active?) segment

**Additional context**
This poses two challenges:
- Segments in WLED are made to be very dynamic. One preset can contain 10 segments and another just 1.
We'd need to assume that the user would want to run with a particular config most of the time, or alternatively add a setting for the number of segments which are controllable by Alexa.
- Naming. Either named segments could be introduced or we could go with a simple continuous numbering `Kitchen 1`, `Kitchen 2`... the user has the ability to change the invocation name in the Alexa app anyways.

ProtoxiDe22, 2020-11-30:
Instead of using segments, it would be possible to expose presets to alexa, as scenes
https://developer.amazon.com/en-US/docs/alexa/smarthome/provide-scenes-in-a-smart-home-skill.html

but that would require having a smart home skill for wled, which is not the approach that is being used right now as far as i understand

disc0018, 2021-11-29:
I use Wled with multipin support. I followed the discussion about the Alexa Light skill.... and the problems with different presets.

For me the solution with multiple Alexa devices mapped to different presets is the most flexible because I can map segments, effects etc. to different Alexa devices.

TheMasterFX, 2022-04-18:
I started to implement this. I'm not quiet sure if this is the right way to do it, or maybe it is better to make a usermod.
You can check my branch: https://github.com/TheMasterFX/WLED/tree/alexa_segment

You can set the option on or off via Webinterface => Config => Sync Interfaces
Each Segment is provided with the segment name to Alexa and can be switched on/off, change color and brightness . The main invocation name controls all segments

kyrenia69, 2023-04-12:
hi , im looking to configure wled on a single  esp32 device to allow alexa to control 6 separate pins ( segments) of single color pwm leds  independently. will themasterfx branch work for this ? thanks

softhack007, 2023-04-15:
I think we have something similar  in WLED already.
* https://github.com/Aircoookie/WLED/pull/2787

kyrenia69, 2023-04-22:
> I think we have something similar in WLED already.
> 
> * [Feature/nine additional alexa devices for presets #2787](https://github.com/Aircoookie/WLED/pull/2787)
thanks so much for the pointer to take a look at the Also emulate devices to call the first  presets

kyrenia69, 2023-04-22:
Ive been playing allday with no success. I  have the presets working fine in wled . alexa is able to find the presets names , but when alexa is requested to run the preset it seems to control all the i/o pins  of my   wled device.
Does this alexa function allow 9 separate i/o pins to be independently controlled by alexa 
im so confused ! is there any instructions on how to set it up ? appreciate any help forth coming  . cheers

softhack007, 2023-09-06:
Implemented with #2787.