wled/WLED #1330: Reset segments with api

DarthWeber, 2020-11-11:
Hi ! Is there any way to delete/reset the segment setup with api commands ?

Aircoookie, 2020-11-11:
Hi!
The JSON API is meant for dealing with segments and highly recommended. To delete them, you just set the stop LED to 0 for every segment but the first:
```json
{"seg":[{},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```

In the HTTP API segments are only supported in a limited way - you'll have to send an API request for each segment you want to delete.
```
SS=<insert segment number here>&S2=0
```

DarthWeber, 2020-11-11:
Thanks - can i somehow send those json commands via mqtt ?

Aircoookie, 2020-11-12:
I was about to say no - supporting the JSON API with the current MQTT library is not easy due to a maximum payload size of about 1400 bytes.
But then I realized that this is only really an issue for the ESP publishing state changes - _receiving_ JSON API payloads should actually work.
So I added support for it into the source right now :)
You'll need to compile that and upload it to your ESP, I recommend Platform.io. Otherwise, a new release should be coming soon :)

closed 2021-06-06 by stale[bot]