wled/WLED #1331: Support for Open Pixel Control

token47, 2020-11-11:
Does WLED support the protocol "Open Pixel Control"? Has it been considered? Seems like it's a better alternative to the E1.31 / DMX one. I'm looking into a project that can control devices over OPC.

The reason I marked this as a feature request is because it seems it's not supported and this can be a request for support.

http://openpixelcontrol.org/

Thanks!

Aircoookie, 2020-11-12:
Not yet, but there are plans to support it at some point :) (see #1241)
In the meantime, the DDP and TPM2.net protocols are a more lightweight alternative to Art-Net and E1.31 that is now supported in the latest WLED source!

closed 2020-11-12 by Aircoookie

token47, 2020-11-12:
Nice, thank you, I'll take a look at those.

And thanks for the good work.