wled/WLED #1333: NodeMCU "Crashing" after Changing Effects

enz1ey, 2020-11-12:
So I have the following components to my permanent outdoor lighting:

(2) 5v/5m/300-pixel WS2812B LED strips (10m/600 LEDs total)
5v/20A/100W PSU
ESP8266 NodeMCU
The strips are connected end-to-end, with power being injected in the middle where the two strips meet. The NodeMCU is at the start of the first strip. I haven't gotten around to soldering yet, so it's just connected with some breadboard jumpers into the JST connector. I was using a 9-meter JST extension cable to get the NodeMCU in my attic and out of the weather, but for troubleshooting purposes, I eliminated that.

Basically what's happening is when I set three colors (red, orange, purple) and cycle through the effects in WLED, after about two or three changes the lights freeze on a solid color/gradient and the NodeMCU goes offline. The blue LED on the chip is off, even though the LEDs are still powered on, and there is a handful of LEDs at the start of the first strip that stay white, regardless of what color the programmed effect was showing. Here is a picture of what that looks like:

[Frozen LEDs](https://i.imgur.com/t86HfvY.jpg)

Would anybody be able to provide some advice? As I mentioned above and you can see in the picture, I eliminated the 9-meter JST extension thinking there might've been too much of a voltage drop across that cable, but the issue persists. I was testing these out in my basement with the lights strung across the floor in this exact configuration, and I don't remember seeing this issue happening then.

gameman733, 2020-11-13:
Where is the power supply in relation to the NodeMCU? If there is a strand of lights between the NodeMCU and the power supply, that might be worth trying to bypass.

enz1ey, 2020-11-13:
The power supply is between the strips with the NodeMCU at the opposite end of a strip. I did try bypassing that and just powering the NodeMCU via the USB port, didn’t fix it.

rchcd, 2020-11-13:
Try to use a logic shifter specially if you want to use an extension cable.

pbolduc, 2020-12-10:
It could be related to #1431

enz1ey, 2020-12-10:
> It could be related to #1431

That might very well be it!

closed 2021-06-16 by stale[bot]