wled/WLED #1335: esp 07 - WLED error 2 ! whit WLED_0.10.2_ESP8266.bin

ddijanic, 2020-11-13:
https://ibb.co/H2RcS38

it works and can change colors and effects.. 
anywhere I click it types error 2
and after power of, and power on, nothing has been saved (wifi setting, led number, animation or color, noting saved.,)

ddijanic, 2020-11-13:
WLED_0.10.2_ESP01_1M_ota.bin
fix my led settings are saved after power off and power on. Color or effect not saved after power off and on.
new problem, works only power PC ip adres, no wlan app on android, and no on phone over ip :(

sansillusion, 2020-11-13:
Wled does not save color and effect on power off. You need to save in a
preset (preferably preset 16 if you want to save segments) and load that
preset at startup.

srg74, 2020-11-13:
> WLED_0.10.2_ESP01_1M_ota.bin
> fix my led settings are saved after power off and power on. Color or effect not saved after power off and on.
> new problem, works only power PC ip adres, no wlan app on android, and no on phone over ip :(

Check out memory size of your module. Looks like it’s only 2M. Default ESP07 is 4M.

Morphy99, 2020-11-16:
> Check out memory size of your module. Looks like it’s only 2M. Default ESP07 is 4M.

Is there a way to check this from WLED? I checked info but it just lists free heap.
I'm having trouble updating from v0.10.00 to v0.10.02 on ESP-07 with the APA102 binary.

srg74, 2020-11-16:
If you uploading with ESPhome flashing tool it will detect your memory size and say 2M or 4M. Error 2 usually means wrong memory size of bin file have been used.

Morphy99, 2020-11-16:
I haven't got serial access to it now so trying to do OTA. It works fine on 0.10 so I'm guessing it's not 2M and it's something else which is stopping the OTA update

closed 2020-11-20 by ddijanic