wled/WLED #1336: ESP32 Bootloop with DMX enabled

davidboon78, 2020-11-13:
Hey, so I'm trying to enable DMX on an ESP32 and it gets stuck in a bootloop, never even gets to initial setup screen.  Is there a better version to use?

 - Board: ESP32 dev (ESP32-WROOM-32)
 - Version WLED v0.10.2
 - Format Self compiled from Arduino IDE

Let me know if there's anything I can add.  I literally only disabled websockets and enabled dmx.

Thank you for your help!

Aircoookie, 2020-11-13:
Hi!
Try a setting a different LED pin in NpbWrapper.h, GPIO 2 is the DMX pin.

We also have a PR that proposes a different DMX library as the one currently used might only work on ESP8266: https://github.com/Aircoookie/WLED/pull/817
However, it is quite dated and would need to be updated significantly in order to work with the current master.

davidboon78, 2020-11-13:
I'm currently using GPIO 12..  is there a recommended pin?  I'll also try that alternate library.  Thanks!

davidboon78, 2020-11-15:
I figured out that i don't need to enable dmx to use artnet from resolume.. sorry to have bothered you. lol thank you for taking the time to respond though.

Aircoookie, 2020-11-16:
Glad you got Art-Net working! Doesn't change anything though about the fact that DMX out seems to be problematic on ESP32, so thank you for reporting :)

davidboon78, 2020-11-16:
I did try out that fork and it worked well, though he includes the library twice so I had to comment one of them out.  

Also found with 1024 leds the framerate was way too low. But I guess thats asking a lot of an esp32.

Aircoookie, 2020-11-16:
The framerate was low in general (effects), with DMX out enabled or with Art-Net?
1024 LEDs are able to achieve exactly 30fps, which is not that bad. However you need 7 DMX universes to transmit that via Art-Net. It would be faster via TPM2.net if your sending software supports that, you'll only need 3 packets with a packet size of 1030 bytes each.

davidboon78, 2020-11-16:
It doesn't seem slow with the effects. Just artnet.  Resolume only supports artnet, does wled support tpm2.net?

Aircoookie, 2020-11-17:
WLED does since 0.10.2, but that won't help you if Resolume only supports Artnet.
In my experience, the maximum "smooth" amount of DMX universes in Art-Net is 3 (= 510 LEDs), sorry about that.

smalljoshua, 2021-01-08:
I'll add that I'm still seeing this bootloop on 0.11.1 on ESP32, I'm not well-enough versed to swap out the libraries as discussed in #817 in the latest release. Any suggestions? I've changed the LED pin to 3 from 2 and kept the DMX pin at 2. I've disabled things like Alexa, Blynk, Hue & IR to try and lighten the load on the processor. If I can do sACN send I could do that instead if anyone has any ideas?

robiness, 2021-02-04:
Hello everybody!
I am also having a bootloop with DMX being enabled on 0.11.1 on ESP32, Without DMX it is working as expected :)

The decoded Stacktrace is
```
PC: 0x400817b1: _uart_isr at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-uart.c line 83
EXCVADDR: 0x00000000

Decoding stack results
0x400817b1: _uart_isr at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-uart.c line 83
0x4008b0ed: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x4011689e: esp_intr_alloc_intrstatus at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c line 666
0x401168f1: esp_intr_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c line 686
0x40112249: uartEnableInterrupt at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-uart.c line 110
0x40112353: uartAttachRx at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-uart.c line 151
0x40112762: uartBegin at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-uart.c line 214
0x4010f100: HardwareSerial::begin(unsigned long, unsigned int, signed char, signed char, bool, unsigned long) at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\HardwareSerial.cpp line 55
0x400f19eb: DMXESPSerial::update() at wled00/src/dependencies/dmx/ESPDMX.cpp line 93
0x400e44cf: handleDMX() at wled00/dmx.cpp line 58
0x400f4b70: WLED::loop() at wled00/wled.cpp line 159
0x400f5556: loop() at D:/Develop/LEDESP32/WLED/WLED/wled00/wled00.ino line 20
0x40112b5d: loopTask(void*) at C:\users\robin\.platformio\packages\framework-arduinoespressif32\cores\esp32\main.cpp line 19
0x40089f81: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

So i guess it's an issue with ESPDMX.  [LXESP32DMX](https://github.com/claudeheintz/LXESP32DMX) does use a modified version of  `esp32-hal-uart.c `.

I am super interested in a working DMX output for this awsome project and i am willing to contribute but i am no c++ developer but i'll see what I can do.

windows10luke, 2021-08-09:
DMX Output is still not working as expected for me as well when using an ESP32.

closed 2021-08-09 by Aircoookie