wled/WLED #1337: Individual Pixels on Universe Boundaries Stay Off When Controlled by E1.31

pierslawson, 2020-11-13:
**Describe the bug**
I have a single ESP8266 controlling 709 GRB pixels in a stip. When using the built in effects, all pixels work great. When I control the lights using xLights over E1.31 three pixels never turn on (pixels 170, 341 and 510).

When I use version 0.9.0-b1 all pixels work fine when controlled by E1.31.

**To Reproduce**
Control more than 1 universe of LEDs using E1.31.

**Expected behavior**
I expected all LEDs to be controlled OK using E1.31.

**WLED version**
 - Board: ESP8266 NodeMCU
 - Version: 0.10.2 Build 2008310
 - Format: Binary

**Additional context**
The problem pixels are on the boundaries of univeses. Is this a division issue?

sansillusion, 2020-11-13:
If I remember correctly there was a bug on dmx miscalculated pixes. I think
it is fixed in the xurrent master 0.10.2

pierslawson, 2020-11-13:
I have 0.10.2 installed at the moment. Do you mean in the master branch but not yet released?

I found this commit https://github.com/Aircoookie/WLED/commit/10c22efbb5decd8bf1f2dd324f7070af46e4608a that I guess you are referring to.

k7bbr, 2020-11-13:
> I found this commit 10c22ef that I guess you are referring to.

Yes. The master branch has the fix for this bug, but the 0.10.2 release doesn’t have the fix. You’ll have to download and compile yourself.

Aircoookie, 2020-11-16:
#1187 :)

closed 2020-11-16 by Aircoookie