wled/WLED #1338: MQTT with OFF message turns ON light if light is OFF

soloam, 2020-11-14:
**Describe the bug**
When the light is OFF and I send a MQTT request with OFF, sometimes the WELD turns on the light and does not accept more MQTT requets. The light stays ON! I noticed this because at night I send a MQTT to all my lights to turn OFF and WLED turns on

**To Reproduce**
Try to get the light off and send several MQTT requests with OFF

**Expected behavior**
Light Stay OFF and never turn ON

**WLED version**
 - Board: ESP32
 - Version 0.10.2
 - Format Binary

Thank you for your help!

huggy-d1, 2020-11-16:
Instead of sending an off command by itself, send instead a DIM or brightness 0 command, then send the OFF command.  This way, when WLED sets itself to off, the strip's last command was to essentially be off.
Are you turning off WLED, or the strip?  Turning off WLED sounds like what you are doing, and and my suggestion should help.

soloam, 2020-11-16:
I'm using the mqtt to turn off wled, then I turn off the power to the strip
(a separate relay). But first I turn off wled, and it's at this time that
the error occurs, if I send more that once the off, the wled turns on and
I'm not able tu turn it off with wled

huggy-d1, 2020-11-16:
Do you mean, send WLED command to turn itself off, like hitting the power button on the WLED interface?
Then do it again after it is already off, and suddenly WLED stops responding and is unable to regain control even with a WLED ON command?

soloam, 2020-11-16:
Yes, I send off by the mqtt, that is equivalent on pressing off in the
interface, and if I send it again via mqtt, wled stops responding via mqtt,
but I'm still able to control it on the web interface! Only the mqtt api
stops responding!

Aircoookie, 2020-11-17:
Thank you for the report and sorry for the issue!
Somehow I can neither reproduce this when sending `OFF`, neither with the 0.10.2 ESP32 binary nor with the latest dev source.
It always turns off as expected, stays off and stays connected to MQTT (a subsequent `ON` after multiple offs will work as expected)

You can give the latest 0.11.0 alpha pre-release a try if you'd like: https://drive.google.com/file/d/1WK-fKmqsg-fb5pqyI8g6VUZ5aW4U8LaX/view?usp=sharing

closed 2021-06-06 by stale[bot]