wled/WLED #1339: WLED App

anrada73, 2020-11-14:
I am using the wled app for another project. You can control the strips via the app easily, but on the homepage it says that LED is offline. Could somebody tell me which parameters in the wled project control this? I want to adjust my project appropriately.

anrada73, 2020-11-14:
https://ibb.co/NnFg2fV
https://ibb.co/jkHM9DW

Aircoookie, 2020-11-15:
Your device needs to implement parts of the WLED HTTP API to work with my app.
On `/win` it needs to return
```xml
<vs>
<ac>128</ac>
<cl>255</cl>
<cl>160</cl>
<cl>0</cl>
</vs>
```

You can replace the AC with your application's brightness value and CL with the set RGB color. If make your application support a `/win&A=xxx` where xxx is the brightness from 0-255 and the command `/win&T=2` that toggles on/off, even the power button and brightness slider will work.

I can't guarantee that it will work though as this app is not designed to work with anything but WLED. Although most of it will work as it is technically just a web browser :)

anrada73, 2020-11-15:
Thanks for the answer.  I already sniffed the traffic yesterday, today it will work. )

anrada73, 2020-11-15:
> Your device needs to implement parts of the WLED HTTP API to work with my app.
> On the `[IP]/win` HTTP endpoint it needs to return
> 
> ```
> <vs>
> <ac>128</ac>
> <cl>255</cl>
> <cl>160</cl>
> <cl>0</cl>
> </vs>
> ```
> 
> You can replace the AC with your application's brightness value and CL with the set RGB color. If make your application support a `/win&A=xxx` where xxx is the brightness from 0-255 and the command `/win&T=2` that toggles on/off, even the power button and brightness slider will work.
> 
> I can't guarantee that it will work though as this app is not designed to work with anything but WLED. Although most of it will work as it is technically just a web browser :)

Done.  Please tell me, what is the principle of device auto-discovery?

anrada73, 2020-11-15:
Screenshots.
https://ibb.co/Sxbb3ZW
https://ibb.co/ZTTZcjd

anrada73, 2020-11-15:
> Your device needs to implement parts of the WLED HTTP API to work with my app.
> On the `[IP]/win` HTTP endpoint it needs to return
>
> ```
> <vs>
> <ac>128</ac>
> <cl>255</cl>
> <cl>160</cl>
> <cl>0</cl>
> </vs> [...]

Screenshot: https://ibb.co/56QrQ16

Aircoookie, 2020-11-15:
The auto discovery is mDNS looking for the `_wled._tcp` service type. Then the app will contact the device over HTTP `/win`, if parsing of the returned XML succeeds the device is added to the list. Since the app automatically detects the name of your device `Ambilight` in the video, I would guess your returned XML is correct and also contains the <ds> property. Not sure what is going wrong here exactly...

Aircoookie, 2020-11-16:
Closing since not an issue or FR. But you are free to ask more question, I'll still get notified on new msgs :)

closed 2020-11-16 by Aircoookie

anrada73, 2020-11-16:
> The auto discovery is mDNS looking for the `_wled._tcp` service type. Then the app will contact the device over HTTP `/win`, if parsing of the returned XML succeeds the device is added to the list. Since the app automatically detects the name of your device `Ambilight` in the video, I would guess your returned XML is correct and also contains the property. Not sure what is going wrong here exactly...

Danke für die Hilfe.