wled/WLED #1340: only 1/3 of 300 leds working

liam922, 2020-11-15:
To be short: I bought a 5m ledstrip with 300 leds, with usb to a 5v adapter. Leds worked fine although one month ago 5 leds suddenly stopped working at the ending. No big deal for me as long as it stayed with those 5.

Yesterday, however. Suddenly almost 2/3 of the leds changed colour and effect randomly. As I wanted to change it back to something else I noticed only about 1/3 of the leds responded when I changed effect/colour. 

After going through settings and some googling I managed to re flash the leds with the latest version 1.02 but I just cant get more than this 1/3 of the leds to work anymore. They just stay dark now although they lit up for a minute with the basic color after flashing them. 

I'm far from an expert with these things. Leds cant possibly be fysically damaged nor did I mess with any settings...

Anyone a clue of what went wrong here? I would love to get the full use of my ledstrip again since now it's only the least visible part in my room that still works... 

Thanks in advance people!

Legsmaniac, 2020-11-15:
How are you powering the led strip? The 5v adaptor? Does it allow at least 10 amps? 300 LEDs required a LOT of power. It's therefore possible your problem is the power supply.
More power, Cap'n!
If it's not the power supply then it's possible it's simply the LEDs are failing, especially if they were a cheap Chinese product.

liam922, 2020-11-15:
@Legsmaniac 
Thank you for the quick reply sir!
The leds are indeed charged with a simple phone charger 5V 0.85a? That's what the seller told me to use... 
Although they might yet be a cheap Chinese product I bought them from a dutch guy who started a small business selling them programmed for 50 euro. It's possible that this is a bit overpriced? I hadn't a clue at the time.

Anyway if I use the 12v adapter from my iPad pro and change the settings it still only 1/3 of the leds that work

Legsmaniac, 2020-11-15:
12v adapter from an ipad pro? Oh my! Under no circumstances use more than 5v on your led strip, you'll blow them up instantly!  And yes .85a is nowhere near enough to power them. Sorry. 

You bought what exactly from a Dutch guy selling them programmed? Both the 300 led roll and a programmed Arduino?

Igor1976-light, 2020-11-15:
You need a 5V power supply of at least 15 A for your 300 LEDs. Your USB power supply is sufficient for the ESP and a few pixels. I am amazed that it even worked

liam922, 2020-11-15:
> 12v adapter from an ipad pro? Oh my! Under no circumstances use more than 5v on your led strip, you'll blow them up instantly! And yes .85a is nowhere near enough to power them. Sorry.
> 
> You bought what exactly from a Dutch guy selling them programmed? Both the 300 led roll and a programmed Arduino?

@Legsmaniac woahh well the 1/3 luckily still works hahaha. I bought both the ledstrip and the programmed arduino box, he didn't give an adaptor tho. He just said to use a 5v one. Phone chargers would suffice. Appears to be not the case...

Website was called Effectled.nl 

You think the leds still can be fixed or are they wasted?

liam922, 2020-11-15:
> You need a 5V power supply of at least 15 A for your 300 LEDs. Your USB power supply is sufficient for the ESP and a few pixels. I am amazed that it even worked

Well it worked for a long time. Anyway the seller should've been a lot more specific to what adaptors you can use ...

Legsmaniac, 2020-11-15:
Most of them are likely to be pretty wasted I'm afraid. You might be able to salvage some by cutting out the blown ones and rejoining the strip, aiming you have a soldering iron and soldering skills, but it's a case of finding which are salvageable and which are fit for the bin.  Personally, I'd just buy a whole new roll.

I would recommend Googling and reading up about addressable led strips. 😉

liam922, 2020-11-15:
Well that's pretty annoying for the € 50 price ticket. 😂

I suppose I will have to. 

How do I get a suitable charger tho? What am I looking for? Since a phone charger clearly doesn't do the job.

Anyways thanks again for the help

Igor1976-light, 2020-11-15:
Here are two examples so that you can see what exactly you need:

[https://www.ebay.com/c/726112650](url)

[https://www.amazon.de/Supply-Netzteil-WS2801-WS2812-WS2812B/dp/B00KKVJ1H2](url)

You need a 5V 20A power supply unit  for the 300 LEDs, a 1000µF 16V capacitor, assemble everything and then you can see how far your LEDs are still shining.
If you need help assembling it, get in touch. You can also find some information in the WLED Wiki.

Igor1976-light, 2020-11-15:
upps... here a Mail from effectled.nl

Dear .....,

I’m sorry for the inconvenience I caused and would like to solve it. I didn’t realise that what I was doing was wrong since wled has the MIT license. I changed the description on my site and mentioned wled and aircoookie with a link to the github page. Is this oke?

Yours sincerely,

liam922, 2020-11-15:
> Here are two examples so that you can see what exactly you need:
>
> [https://www.ebay.com/c/726112650](url)
>
> [https://www.amazon.de/Supply-Netzteil-WS2801-WS2812-WS2812B/dp/B00KKVJ1H2](url)
>
> You need [...]

Hmm thank you @Igor1976-light! I will have a look into this although the ledstrip has to be powered with an adapter with a USB gate. 

However I got a reply from the seller and he's gonna send me a whole new ledstrip. I've got to mention that the service has been really good with this guy! 

Anyways still gonna look into it to get the right adapter for it.

liam922, 2020-11-15:
> upps... here a Mail from effectled.nl
>
> Dear .....,
>
> I’m sorry for the inconvenience I caused and would like to solve it. I didn’t realise that what I was doing was wrong since wled has the MIT [...]

Ohhh hahah I hope everything's ok for his shop.
Didn't mean to mess with his stuff

Igor1976-light, 2020-11-15:
As already mentioned, nothing works with a USB power supply. The USB adapter is only a connection for the power supply. I could also use a 3 pin XLR or any other plug (1x -, 1x +) You NEED a proper power supply for the LEDs (300 LEds * 60mA = 18000ma / 18A)

Igor1976-light, 2020-11-15:
äää.... 18000mA=18A

Igor1976-light, 2020-11-15:
> Ohhh hahah I hope everything's ok for his shop.
> Didn't mean to mess with his stuff

It's just a cheek to pass this off as his program. But also sell 5m LEDs without referring to the required power supply units.

liam922, 2020-11-15:
@Igor1976-light oh but honestly that's all chinese for me...

My apologies, 

Although if you look at the site of the seller there's this little black box that connects the ledstrip with the power supply. Could this be what you meant or is it still something else?

Kind regards

liam922, 2020-11-15:
> It's just a cheek to pass this off as his program. But also sell 5m LEDs without referring to the required power supply units.

Ah so it is something extra then!

Igor1976-light, 2020-11-15:
I looked at it on the side of me, and can only pull my hair out over the madness.
I also bet if I connect the USB cable 20A it will melt because the cable cross section is too small.

Send it back. But if you want to keep it, there is no way around opening the box and lending a hand yourself.
But we are happy to help you, provided you know how to solder or know someone in your circle of friends.

liam922, 2020-11-15:
@Igor1976-light thanks for your contribution my friend! As I don't really know someone, and I'm not really capable of doing it myself without spending money on certain stuff ... I'll just wait it out and see if the leds get messed up again, which has a big chance to it as they're lit up for like 4 hours a day. So if it happens again i'll just ask my money back... 

Once again you have my thanks!

Igor1976-light, 2020-11-15:
Since your problem is more hardware-related and has power and nothing to do with code, could you please remove the etiquette bug?
 It is confusing.

liam922, 2020-11-15:
> Since your problem is more hardware-related and has power and nothing to do with code, could you please remove the etiquette bug?
> It is confusing.

It seems like I'm not able to edit anything more than the title of the issue.

Apologies for that!

If you however can explain me how it can be done i'm definitely willing to do it.

pbolduc, 2020-12-10:
Can this just be closed?

Aircoookie, 2020-12-10:
Closing for now. Feel free to post additional comments if the issue is not yet resolved.

closed 2020-12-10 by Aircoookie