wled/WLED #1342: WS2812B (5m 300 LEDs) the first 60 LEDs do not work

mluckau, 2020-11-15:
**Describe the bug**
The first 60 LEDs do not work correctly. Only 7 LEDs light up at irregular intervals. I tried this with 2 different LED strips, both show the same behaviour. (I had shortened the first one, because I thought that the first LEDs are defective, but after shortening the first 60 LEDs are off again, which worked fine before).

**To Reproduce**
Flashing ESP32 with the current version and connect 300LED WS2812b strip

**Expected behavior**
All 300 LEDs are lit

**WLED version**
 - Board: ESP32
 - Version 0.10.2
 - Format Binary and self-compiled]

Igor1976-light, 2020-11-15:
If you have already cut off the first 60 LEDs, then just try these.

mluckau, 2020-11-15:
yes, I have already done that, but then again the first 60 of the 240 do not work. i have also tried it with another led stripe, the same thing happens.

Igor1976-light, 2020-11-15:
Connect the 60 LEDs times. Or have you already tried this?

liam922, 2020-11-15:
> **Describe the bug**
> The first 60 LEDs do not work correctly. Only 7 LEDs light up at irregular intervals. I tried this with 2 different LED strips, both show the same behaviour. (I had shortened [...]

Sounds familiar, I created an issue about the same thing not long ago. You might find it interesting to read. 
https://github.com/Aircoookie/WLED/issues/1340 
Here's the link to it.

Igor1976-light, 2020-11-15:
@mluckau What kind of power supply do you use? I mean how many amps does your power supply have?

mluckau, 2020-11-15:
@Igor1976-light 5V 26A 
https://www.amazon.de/WITTKOWARE-HRPG-150-5-Netzteil-Industrie-130W/dp/B08KD2YL4X/

Igor1976-light, 2020-11-15:
Did you hook it up after the wiki? 

[https://github.com/Aircoookie/WLED/wiki](url)

Without level shifter. Power supply directly from the power supply to the LEDs and the data from the ESP to the LEDs?

Igor1976-light, 2020-11-15:
> @Igor1976-light 5V 26A
> https://www.amazon.de/WITTKOWARE-HRPG-150-5-Netzteil-Industrie-130W/dp/B08KD2YL4X/

Der Preis ist ja Wucher

mluckau, 2020-11-15:
@Igor1976-light den Preis habe ich auch nicht bezahlt habe auf die schnelle nur ein link zu diesem Netzteil rausgesucht. Habe glaube ich damals auch nur ca. 30€ dafür bezahlt und es ist sogar das von MEAN WELL. 

Anfang und Ende von dem LED Streifen habe ich direkt an das Netzteil angeschlossen und die Datenleitung an Pin 02 über ein 470 Ohm widerstand an den ESP32. Der ESP32 wird über USB versorgt und GND von Netzteil und ESP32 habe ich verbunden.

English:
I connected the beginning and end of the LED strip directly to the power supply and the data line at pin 02 over a 470 Ohm resistor to the ESP32. The ESP32 is powered via USB and I connected GND of the power supply and ESP32.

mluckau, 2020-11-15:
@Igor1976-light hier gibts das 200W sogar unter 30€ ;-)
https://de.aliexpress.com/item/4000367052959.html

Igor1976-light, 2020-11-15:
You have to supply both with power from a power supply unit. With two power supplies there are two separate circuits. In the worst case, by connecting GND you break your PC. (If the USB connection runs over it)

Du musst beides von einem Netzteil mit Strom versorgen. Es sind mit zwei Netzteilen so zwei getrennte Stromkreise. Mit verbinden von GND machst du im schlimmsten Fall deinen PC kaputt. (Sofern der USB-Anschluss darüber läuft)

mluckau, 2020-11-15:
Ok, according to wiki I understood that it should work like this. With the 12V variant one does it also in such a way. But well, I changed it now, I don't have a 1000uF capacitor there, but it should work like this. But it did not change the behaviour. The first meter is still crazy.

Ok, habe es laut Wiki eigentlich so verstanden das es auch so gehen sollte. Bei der 12V Variante macht man es ja auch so. Aber gut, habe es jetzt geändert, habe zwar gerade keinen 1000uF Kondensator da, aber es sollte ja auch so gehen. Es hat aber nichts am Verhalten geändert. Der erste Meter spinnt immernoch rum.

Igor1976-light, 2020-11-15:
hhhm... Anfangs hatte ich auch Probleme mit langen Strecken von LEDs und konnte sie mit dem Kondensator beheben. Weiß aber jetzt nicht mehr, was die fehlerhaftem Auswirkungen waren.
Die ersten LEDs sind aber nicht defekt. Sonst ginge das Datensignal nicht durch und alle LEDs danach würden nicht mehr Leuchten. Nicht umgedreht.
Die Grafik für die 12 V ist recht unglücklich. !2V Netzteil zum 5V Spannungsregler zum ESP, und von 12V Netzteil zu den LEDs und GND ist somit eine Leitung.

ENGLISH:
hhhm ... At first I also had problems with long stretches of LEDs and was able to fix them with the capacitor. But now I don't know what the erroneous effects were.
The first LEDs are not defective. Otherwise the data signal would not go through and all LEDs would no longer light up. Not upside down.
The graphics for the 12 V are quite unfortunate. ! 2V power supply to the 5V voltage regulator to the ESP, and from the 12V power supply to the LEDs and GND is one line.

Igor1976-light, 2020-11-15:
Der Elko ist dafür da um Energiespitzen zu unterbinden und um variierenden Energieverbrauch kurzfristig zu kompensieren. Je länger der LED Streifen so höher der Energieverbrauch. Auf deinen Streifen hast du auch lauter Kondensatoren drauf, die ebenfals dafür sorgen, das jeder Chip läuft.

The electrolytic capacitor is there to prevent energy peaks and to compensate for fluctuating energy consumption at short notice. The longer the LED strip, the higher the energy consumption. You also have a lot of capacitors on your strips, which also ensure that every chip runs.

mluckau, 2020-11-15:
Ich habe mir Kondensatoren bestellt, die sollten morgen ankommen, werde es dann damit probieren. In der Zwischenzeit teste ich nochmal ob ich den LED Streifen einfach über die Neopixel Library am ESP komplett benutzen kann(Also ohne WLED). 

English:
I have ordered capacitors, they should arrive tomorrow, I will try it with them. In the meantime I'm testing again if I can use the LED strip completely via the neopixel library on the ESP (without WLED).

Igor1976-light, 2020-11-15:
Besorg dir einen 1000µF 16V Elektrolyt-Kondensator und teste es noch einmal. Ich hab WLED auf NodeMCU, ESP32 und ESP32_POE ohne Probleme am laufen.

Get a 1000µF 16V electrolytic capacitor and test it again. I have WLED running on NodeMCU, ESP32 and ESP32_POE without any problems.

mluckau, 2020-11-15:
Ok, der Kondensator sollte morgen dann ja da sein, werde es ausprobieren und hier dann berichten. 
Vielen Dank für deine Hilfe.

Ok, the condenser should be there tomorrow, I will try it out and report here.
Many thanks for your help.

Igor1976-light, 2020-11-15:
sorry, zu spät gelesen

mluckau, 2020-11-15:
Mir ist gerade aufgefallen, das selbst im Webinterface der erste Meter als nicht vollständig eingeschaltet angezeigt wird, also scheint sich der LED-Streifen tatsächlich so zu verhalten wie es die Software ihm sagt. Ist es eventuell doch ein Softwareproblem?

I just noticed that even in the web interface, the first meter is shown as not fully on, so the LED strip actually seems to behave as the software tells it to. Is it a software problem after all?

[image]

Igor1976-light, 2020-11-15:
ckeck it without LES, but first reboot.

mluckau, 2020-11-15:
Habe jetzt die Spannungsversorgung ausgeschaltet dann habe ich den Widerstand entfernt (also die Datenleitung zu den LED Streifen unterbrochen) und die Spannungsversorgung wieder eingeschaltet.
Auf dem Webinterface sieht es genauso wie vorher aus, der erste Meter ist weiterhin nicht komplett eingeschaltet.

If I switched off the power supply then I removed the resistor (i.e. interrupted the data line to the LED strips) and switched on the power supply again.
On the webinterface it looks the same as before, the first meter is still not completely on.

[image]

Aircoookie, 2020-11-16:
Danke für die Meldung! Da das Problem auch in der Vorschau auftritt, muss es sich um ein Softwareproblem handeln.
Versuch doch bitte einmal den Flash komplett zu löschen (z.B. mit `esptool erase_flash`) und dann die Binary neu zu flashen. Das hat bei einem ähnlichen Problem schon einmal geholfen.

Thank you for reporting! I suspect it's a software problem, as the problem is also visible in the preview.
Please try erasing the flash completely (e.g. with `esptool erase_flash`) and then flash the binary again. That has helped with a similar issue in the past.

Isn't it weird to write the same comment both in German and English :)

mluckau, 2020-11-16:
Hallo @Aircoookie und @Igor1976-light ,

danke für den Tipp.
Ich habe den ESP32 mit esptool.py komplett gelöscht. Anschliessend habe ich mit esptool.py die WLED_0.10.2_ESP32.bin neu geflasht und jetzt funktioniert es tatsächlich wie erwartet.
Vielen Dank für die Hilfe. 
Ich hätte nicht gedacht das dass Problem so behoben werden kann. Das ist ja schon recht merkwürdig.

Hello @Aircoookie and @Igor1976-light ,

thanks for the tip.
I have completely deleted the ESP32 with esptool.py Afterwards I flashed the WLED_0.10.2_ESP32.bin with esptool.py. Now it really works as expected.
Thanks for the help. 
I didn't think the problem could be solved this way. This is quite strange.

closed 2020-11-17 by Aircoookie