wled/WLED #1344: Random modes

themanfrommoon, 2020-11-16:
Hello, thank you for the great software!
I have some ideas:
Please add a lot of random modes:
-Random choose of the palettes
-Random choose of the effects
-Random choose of the colors
-Choose if random alternation should be done on every reboot or after a (random) time
-Choose if the palettes can change during the same effect or if palettes and effects are change at the same time
-Change random palettes during the same effect
-you can random change everything
....and it will never be boring because of the same effects at any time
Thank you and best regards,
Chris

pbolduc, 2020-11-17:
I plan on doing something similar to this, however, I'm going to use node-red to control wled via the http interface commands. I can understand wanting it all included but that will take take time and probably wont be available for my lights this year. Plus I want more integration so when I turn my WLED on for outdoor, I want to turn off the house lights and when I turn off the WLED at 10 or 11pm, I want my house lights to turn on.  So node red will work for my use case.

justusiv, 2020-11-30:
I used this app from the play store to accomplish this via http https://play.google.com/store/apps/details?id=ch.rmy.android.http_shortcuts&hl=en_US&gl=US 
Although i probably wouldn't mind seeing a random at the top of each list myself. We have on on the color selection.

Zebble, 2020-12-21:
For anyone interested in doing this in HomeAssistant, you can add a RESTful sensor to randomly pick an effect:

`sensor: 
  - platform: rest
    name: WLED Randomizer
    resource_template: http://IP ADDRESS OF WLED/win&FX={{range(0,184)|random}}
    scan_interval: 180
    value_template: 'OK'`

You can change the scan_interval to how often you want to change effects (in seconds).  The random range (0 to 184 in this case) can also be adjusted and will need to be adjusted when new affects are added.  In 0.11.1 there are currently 185 effects (0 to 184).

andrew-moiseyev, 2022-02-09:
So currently no random in plain WLED — without buttons, HA, HTTP, JAON, etc.?

blazoncek, 2022-02-09:
you can use `{...{..."fx":"r",...}}`and equivalent for random values in JSON API.
equivalents exist (but not all) for HTTP API

andrew-moiseyev, 2022-02-09:
Thanks, but I mean just standalone board, so we plug in — and it starts showing random effects/colors/times as requested in the first message. Initial configuring — and no further web/app/HA — just board and leds.

blazoncek, 2022-02-09:
So? Save a preset and schedule it (perhaps in playlist).

andrew-moiseyev, 2022-02-09:
Preset "pal": "r", "fx": "r". Playlist with only this preset, with needed duration and indefinite cycle. Am I right?

closed 2022-02-09 by blazoncek