wled/WLED #1345: Add Matrix and Ring LED Arragement & Effects

themanfrommoon, 2020-11-16:
Hello,
thanks a lot for this great software!
I have some ideas for you:
Please add support for Matrix and Ring LED Arrangements & Effects
e.g. Rings: https://de.aliexpress.com/item/32809169128.html
e.g. Matrix: https://de.aliexpress.com/item/4000547842744.html
[image]
[image]

Thank you and best regards,
Chris

huggy-d1, 2020-11-16:
Using 9 segments, you can do that now.
`
Segment   start   end       # 
     1            1        2          1
     2            2       10         8
     3           10      22        12
     4           22      38        16
     5           38      62        24  (I stopped counting here)
     6           62     90?       28?
     7           90?   122?     32? 
     8         122?   158?     36?
     9         158?   198?     40?
`       
Then setup segments to act as you like in each ring.
I agree, having grids, whether in 2-D or 3-D shapes would be a fantastic WLED enhancement.

There is open source github code available for handling specific shapes that could be integrated into a usermod or group of usermods, and perhaps eventually incorporated into the main WLED code.

Have you looked around for existing code that handles your specific device in a way that suits your particular desires?

Aircoookie, 2020-11-17:
The awesome [Sound reactive fork](https://github.com/atuline/WLED) has a few very nice 2D effects 👍 

I want to incorporate that capability into master too at some point, though I can't give a timeline yet.

jlaitinen81, 2020-12-21:
Yes please! The ability to display 2D effects, pixel graphics and text on led matrixes would be so cool!

Kamal-Sonani, 2020-12-22:
Upload any bitmap on web and decode with Raw RGB value & scroll that rgb with image .

eecue, 2021-05-08:
This would be amazing. I scored a bunch of grids from AliExpress. Would love to use as a display even.

blazoncek, 2023-01-25:
Matrix support (and 2D effects) available in 0.14 and up.
For circular/ring LED strips use multiple segments.

closed 2023-01-25 by blazoncek