wled/WLED #1346: Add MQTTS

heini66, 2020-11-16:
**Is your feature request related to a problem? Please describe.**
MQTT isn't secure.

**Describe the solution you'd like**
Add MQTTS and the possibility to apply ssl credentials eg ssl fingerprint

**Describe alternatives you've considered**
There is no alternative to encryption..

huggy-d1, 2020-11-16:
This is an ESP32-only request - at the moment.  Perhaps eventually it will be a raspberry pi request, or some other microcontroller based system with enough processing power and memory to handle certificates AND certificate management.  Since certs have an end date, they need to get updated in a secure way as well.  All these things drive up the complexity and hardware requirements.

ESP8266 is not capable of handling https/mqtts, and I think most WLED systems are running on ESP8266's, so this request while important in certain applications is important, there are other ways to enhance security.

Aircoookie, 2020-11-17:
@huggy-d1 is correct. I've attempted making a TLS connection on an ESP8266 and it only works if literally nothing else runs on the system as there is no hardware accelaration and way too little free RAM.

On an ESP32 supporting secure MQTT is possible though  - cert management remains an issue though. Usually arduino sketches just use a hardcoded fingerprint for the specific server they want to contact, which frankly sucks because it expires after a year or so and only is valid for that specific server. The alternative is not verifying the server signature at all, which gives you encryption, however is of limited use since you can't verify the authenticity of the server you are connecting to.

I'm very keen if someone has good ideas on this point :)

heini66, 2020-11-17:
mqtts as client on an esp8266 is possible .
have a look @ https://github.com/fablab-ka/co2-ampel/blob/master/Software/co2ampel/data/config.json
i would like to have the option to add mqtts credentials in wled to connect to my mqtts broker

Aircoookie, 2020-11-17:
Of course it is possible, just not with the overhead of hosting a full embedded webserver, filesystem and driving potentially more than 1k individually addressable LEDs. And if it barely works, it would drastically reduce stability.

Security should of course be a given in 2020, but its only possible when there are enough resources. Given that no sensitive data is transmitted, stability is more important for this project at this point.

I would however accept a PR that adds it optionally with a compile time define :)

huggy-d1, 2020-11-17:
> mqtts as client on an esp8266 is possible .
> have a look @ https://github.com/fablab-ka/co2-ampel/blob/master/Software/co2ampel/data/config.json
> i would like to have the option to add mqtts credentials in wled to connect to my mqtts broker
I just checked your link.  They cheated.  Hard-coded the SSL fingerprint into the firmware.  It is not at all what you think it is, and in a short period of time, your MQTTS server will quite accepting data from an esp8266 using this software.
That said, if your desire to use WLED is seasonal, that solution should work fine.

RoganDawes, 2020-11-29:
I think there are two possible options that may still be feasible.
1) TLS-PSK (a la mosquitto) - not entirely sure whether there are any libraries that support this on any modern platform, unfortunately.
2) saving the fingerprint of the server's public key (or even the entire public key), not the entire certificate. If you keep the same private and public keypair across certificate renewals, even though the validity period changes, and therefore the signature changes, and thus the fingerprint of the certificate as a whole, you are not giving up a whole lot of security in the process.

huggy-d1, 2020-11-30:
Creating a self-signed certificate manually so you can use the same key fingerprint for extended periods of time?

lhurt, 2022-10-05:
Although this request is quite old Tasmota was able to solve this issue and [support MQTT TLS](https://tasmota.github.io/docs/TLS/) since last year using bear ssl library. Any chance to add it to WLED as well?

lhurt, 2022-12-04:
Is anyone watching this issue?

Aircoookie, 2022-12-04:
Yes. It's possible on ESP32, but not currently a priority. PR is welcome :)

lhurt, 2022-12-05:
> Yes. It's possible on ESP32, but not currently a priority. PR is welcome :)

Thank you for explaining. I'd love to contribute, unfortunately it exceeds my capabilities.
BTW: Tasmota offers [MQTTS also for ESP8266](https://tasmota.github.io/docs/TLS/#compiling-tls-for-esp8266).

Thanks btw for your great piece of software.

max-koehler, 2023-07-04:
Generally speaking, ESP8266 and MQTT TLS can work like a charm. 

The problem is usually establishing the connection, where often out memory issues occur and space issues due to the root certificates that need to be shipped with the firmware. 

I can see that `BearSSL` is already part of the build, and maybe we can scope the feature so that it's easier to start a PR that goes in the right direction.
- Do we have a trust anchors / root certificates that are shipped with `WLED` and loaded into the client already? 
- If not, could we focus on `letsencrypt` only, to safe space and memory?

blazoncek, 2023-07-05:
> Generally speaking, ESP8266 and MQTT TLS can work like a charm.

That very much depends on the amount of free RAM and available CPU cycles. With WLED installed ESP8266 struggles with both.

max-koehler, 2023-07-05:
> > Generally speaking, ESP8266 and MQTT TLS can work like a charm.
> 
> That very much depends on the amount of free RAM and available CPU cycles. With WLED installed ESP8266 struggles with both.

In your opinion, it isn't even worth looking into it?

blazoncek, 2023-07-05:
If your life depends on MQTTS then it may be time to switch to ESP32.
With ESP32 things are more flexible although we are already pushing the limits there too.

MateuszPrasal, 2025-11-20:
https://github.com/MateuszPrasal/WLED/tree/mqtts - this is the branch with MQTTS from current `main` branch -  feel free to test :) 

Now, I'm currently do tests on `v0.15.1` tag with 140 LEDs and MQTTS enabled and what I see is that `Free heap` is about 72kB

[image]

I have also disabled lots of things like websockets, alexa and so on. I'll also test on current main branch so I'll let you know about tests