wled/WLED #1347: WLED as driver for 5mm warm white LEDs?

Joshfindit, 2020-11-16:
Ok, so I understand that WLED is entirely overkill for what I want, but honestly the big pull is being able to use Alexa and the App

Here's what we have:

ESP32s (cause the 25kHz+ LED refresh rate is 🙌)
1-5 "segments" of plain LEDs.
There may be "fairy lights", 3mm LEDs, 5mm LEDs, and so-on on any given segment
When there are hundreds of them, I drive them separately and use GPIO pins connected to transistors. When there are a few, I drive them directly from the GPIO pins.

Ideally, I want to be able to have each of these segments be controlled independently and maybe have effects that play off each other (like twinkle, flicker, etc).

I have been doing something passable with ESPHome, but there's no app and no Alexa.

Is this possible with WLED? Am I barking up the wrong tree?

Thanks

Aircoookie, 2020-11-17:
Hi!
There is support for LEDC PWM out in WLED with the `WLED_ENABLE_ANALOG_LEDS` define in NpbWrapper.h :)
It presently supports 4 channels (meant for RGBW, respectively), and they are set to the same color the first digital LED is set to.

In the upcoming 0.12.0 release (hopefully still this year!) I will add multistrip support, and that will be capable of outputting PWM signals for regular LEDs for up to 8 independant channels!

ulte, 2020-11-17:
wait if we use esp32 we get better led refresh rates?!?

Aircoookie, 2020-11-17:
@ulte i believe @Joshfindit is referring to the PWM frequency for traditional LEDs, not the refresh rate of digital LEDs (which always run at a protocol speed of 800kHz for Neopixels)

Joshfindit, 2020-11-17:
> wait if we use esp32 we get better led refresh rates?!?
> @ulte i believe @Joshfindit is referring to the PWM frequency for traditional LEDs, not the refresh rate of digital LEDs (which always run at a protocol speed of 800kHz for Neopixels)

Correct: I’m talking about the PWM frequency of traditional LEDs. the ESP32 has a built in PWM driver that can do up to 40Mhz @1bit, or ~300kHz @8bit.
I’m very sensitive to PWM flicker, and have been very happy with the results of dimming traditional LEDs with the ESP32. Even at the lowest brightness I can’t detect any flicker.

Joshfindit, 2020-11-17:
> Hi!
> There is support for LEDC PWM out in WLED with the `WLED_ENABLE_ANALOG_LEDS` define in NpbWrapper.h :)
> It presently supports 4 channels (meant for RGBW, respectively), and they are set to the [...]

@Aircoookie Thank you for the reply. I did see WLED_ENABLE_ANALOG_LEDS, but I must say I’m confused about how it works.
Does that mean that it creates a “single” light with 4 channels, each channel being a GPIO PWM output? If so: where can the PWM frequency be set (or what is it?), and how do I configure the settings on the webpage?

Aircoookie, 2020-11-17:
Yes, 4 PWM outputs with one each for the Red, Green, Blue and White channel (of an RGBW light). The frequency is the the second argument of the `ledcSetup();` method, which starts at line 225 of NpbWrapper.h. It is set to 5kHz by default.

An easy way to control these channels individually right now is to enable the RGB sliders in UI settings, which will let you set each channel independently.

Joshfindit, 2020-11-18:
@Aircoookie Got that working, 25kHz no problem.
Thank you 🙌

Can't wait for 8 independant channels!
P.s. Watch out for "4 pwm timers are available, multiple channels (up to 8? 16?) can use the same timer" ([https://forum.micropython.org/](https://forum.micropython.org/viewtopic.php?f=18&t=3717&sid=9ff06dba29db7bf8424c88a4b9b2b860&start=10))

Joshfindit, 2021-03-20:
Definitely still interested in this

Joshfindit, 2021-07-21:
Still interested in this

Joshfindit, 2021-09-13:
@Aircoookie I'm happy to report that WLED `v0.13.0-b2 "Toki"` seems to have support for driving 5mm warm white LEDs:

1. Flashed the bin to an ESP32 "mini" (similar form factor to the D1 Mini)
2. Connected warm-white LEDs to each pin (ground was common among all LEDs and the ESP32):
    1. A small string of 3.3v "Fairy lights" to pin `17`
    2. An array of 5mm LEDs to pin `22` (with the appropriate resistor)
    3. An array of 5mm LEDs to pin `23` (with the appropriate resistor)
3. Went to LED setup, and added 3 LEDs:
    1. `PWM White`, Color Order `RGB`, Pin `17`, Index `0`
    2. `PWM White`, Color Order `RGB`, Pin `22`, Index 1`
    3. `PWM White`, Color Order `RGB`, Pin `23`, Index `2`
4. Applied settings and rebooted
5. Turned on the light, and it worked! Was also able to dim and play with effects!

There was an issue (colours constantly flickering wildly), but I'm filing that separately if I can't sort it out.

Joshfindit, 2021-09-13:
(Mainly adding those notes both to help anyone else who is doing the same, and to confirm that that is the supported/acceptable way to do it in the first place)

blazoncek, 2021-09-13:
This functionality is available since April so nothing new here.
Color order is only important for RGB(W) analog LEDs and pin will depend on the actual HW used.
Index is an interesting thing which should follow led count from previous strips and not start at 0.

Joshfindit, 2022-04-16:
I’m comfortable closing this as resolved. The flickering is still an issue, but WLED does currently support analog LEDs out of the box.

Thanks!

closed 2022-04-16 by Joshfindit