wled/WLED #1348: Multiple outputs from 1 esp8266

RavD666, 2020-11-16:
Hi
Is it possible to have 2 strips working from 1 esp8266 
Ideally having 2 gpio pin

Thank you

huggy-d1, 2020-11-16:
If you visit QuinLED.info and search for the ESP32 based WLED alpha version that uses 4 pins to control 4 strips, you can modify that version to work on an esp8266.

Short of that, you can use a 4 input level shifter on ESP8266 data pin and drive 1 "long" or 2 "short" LED strips from a single data pin.
In the 1 data pin, 1 long strip, it would be made up of 2 segments, with segment 1 being one strip, and segment 2 being the other strip.
In the 1 data pin, 2 short strips, 1 data pin goes to both strips.  This results in both strips displaying the exact same thing.
Is that useful information?  

FYI, this is not really an "issue" for WLED.  It is probably best handled in the WLED discourse site at https://wled.discourse.group/

Aircoookie, 2020-11-17:
#104 :)

closed 2020-11-17 by Aircoookie