wled/WLED #1351: mega tree xlights esp8266

jazzvivi2, 2020-11-17:
Hello I want to build a mega tree with about 2000 pixels using several esp8266 due to the channel limitation. there is a way for each of them to start where the previous one ends. In the xlights configuration I don't know how to configure or if the tree can be configured with several controllers as if it were one. The idea is that several synchronized mcu work as if it were one, each with its 7 universes. thank you and congratulations for this project

Aircoookie, 2020-11-17:
Hi!
You can just use multiple controllers (currently I recommend one controller per 500 pixels) and xLights will automatically map them to consecutive channels once you add them :) It is made for these sorts of huge setups after all.

If you were asking about WLED effects, those can't yet be mapped continuously across multiple controllers, only synced. Sorry about that, I am planning to make that possible too at some point!

In a few days, I will release WLED v0.11, which also features DDP protocol support. You could consider using that as it has less overhead than E1.31 and might yield a smoother animation and/or allow you to run 1k pixels per controller.

pbolduc, 2020-11-17:
It takes 30us per pixel to send the data. Once you get too many pixels, it will start taking too long to send the data. At 750 pixels, it takes 22.55 milliseconds to send the data. This is why it is recommended to use multiple controllers.

Example of sending 5 pixels worth of data.

[image]
[image]
[image]

jazzvivi2, 2020-11-21:
thanks for the answers, i'm doing some testing with a couple esp8266s and they work fine. wanting to try that version 0.11 and the DDP protocol for what I read about it I find it very interesting. I never thought that artnet would occupy so many resources in a network. in my theater testing a couple of 512 channel DMX universes with wifi knocks the network down and suffers delays

closed 2021-06-04 by stale[bot]