wled/WLED #1357: WS2812B

lodv, 2020-11-19:
Hi,
Maybe it's a stupid question, but I have several LED 2812B strings.
One has 3 wires (grnd, 5V and green - data), one has 5 wires (black, green, red, blue, white -12V needed) and one has 4 wires (black, green, red, blue - 5V needed. On this string it says gnd - Si - Ck - 5V). I want to use WLED. It works perfect with the first one with the 3 wires on a NodeMCU. But I have no idea how I must attach the other 2 to NodeMCU in order to work with WLED. Is there a tutorial where I can learn how to use them?
Thank you for your answer.
Yvonne

bharvey88, 2020-11-19:
check out a video on youtube. the red and black wires can be used to power nodeMCU and strip instead of using usb.

lodv, 2020-11-19:
Thank you for your help.
But on all those videos I only see 1 data wire (green) and the instruction in the wiki of WLED says that you use D4, so what to do with the blue and red wire (RGB)?
I like to use WLED as I can integrate it in Home Assistant.
Thanks.

bharvey88, 2020-11-19:
that is definitely easy to do! i suggest joining the WLED discord- much easier to get help there.

https://discord.gg/7dEgJ6A44C

huggy-d1, 2020-11-19:
Did you review the YT video?  I believe what they are saying is those 2 "extra" wires can be used to provide power to the NodeMCU.
I would power up the LED strip, connect the datapin, and while the strip was operating, measure the 2 wires to verify red is 5V and black of 0V/GND compared to the 0V and 5V powering the strip itself.
If that is the case, you can power the NodeMCU with those 2 wires and not have to power it from some other set of wires.
You can also simply heat-shrink the ends of the wires and ignore them.

wessi86, 2020-11-19:
The LED strip with 4 wires is more likely a WS2801 and you can use it as well. I'm not sure if the 12V one is supported though.
Connect the WS2801 like this and also use the appropriate firmware:
CK  ---->  GPIO2 (D4)
SI  ----->  GPIO3 (RX)

Good luck,
Christian

sansillusion, 2020-11-20:
For the 12V one you will most likely need level shifters for the communication lines.

lodv, 2020-11-20:
Thank you all for your comments and suggestions. I will continue working on it. 
I hope I succeed. If not, I will open a new issue.
For now, regards and thanks.
Yvonne

huggy-d1, 2020-11-20:
I think the right place for your questions is here: https://wled.discourse.group/
I do not think you have an ISSUE with WLED.  You have questions identifying the LED strips you have and how to wire up and control them.  I think that's correct.
The best source for fast answers is probably the WLED discord server.

lodv, 2020-11-21:
Yes, that's right. Thank you. I have a look over there.

lodv, 2020-11-23:
Christian, 
You mentioned in your answer according the ws2801 ledstrip: use the right firmware. I attached the string to a nodemcu and upload the wled for esp2866 binfile. Is that the right version?
Thank you.
Regards,
Yvonne

closed 2020-11-26 by lodv