wled/WLED #1358: Why does my led light up only 1 light?

danieely01, 2020-11-21:
Hey everyone!
I got an issue with my LED setup.
(I ordered all of these parts: https://tynick.com/blog/11-03-2019/getting-started-with-wled-on-esp8266)
I connected to the WI-Fi, but when I connect my ESP8266 to my LED Strip (WS2812B LED Strip) with jump wires, I see nothing, but only the first led light up or no light at all. I plugged the wires to the right places (Power - Vin, Ground - GND and Digital out to the D4.)

Thank you for your help!

StefanRu1, 2020-11-21:
Hi, did you connect D4 to DIN of the stripe (not DOUT).
Also check that on the config page of WLED -> LED Preferences the LED count is set to the number of LEDs you have on the stripe.
Also take care that when only connected via USB power supply the Power Limiter is set to on.

BR,
Stefan

danieely01, 2020-11-21:
Wow, thank you so much. I connected the strip the wrong end!!
"Also take care that when only connected via USB power supply the Power Limiter is set to on."
Can you explain me this please?
Should I buy a power supply for it?

StefanRu1, 2020-11-21:
Yes, i know this error. 
In my very first project with WLED i had the same error :-)

How many LEDs do you want to drive. 
With only USB you get a problem at i think 1A.
You can destroy your ESP  when having more power running through it.
1A is somewhere between 10 to 20 LEDs.
I think a warning is also written on the settings page on the WLED UI.

So yes for more than a few LEDs you should use an external power supply.
You can power the ESP with it and also connect + and - from the LED stripe directly to + and - on the power supply.
Only having D4 to DIN from ESP to the LEDs.

See also:
https://github.com/Aircoookie/WLED/wiki

BR,
Stefan

closed 2021-06-04 by stale[bot]