wled/WLED #1361: Preset Errors: "preset does not exist" and "enter your API command"

twlare, 2020-11-22:
**Describe the bug**
Errors "The requested preset does not exist" and "Please enter your API command first" when trying to modify a saved preset.

**To Reproduce**
Create a preset and save it. Create a second preset and save it. Open the first preset and try to modify it.

**Expected behavior**
Since the preset was created and saved, I didn't expect "preset does not exist" or "enter your API command" errors.

**WLED version**
 - Board: Wemos D1
 - Version 0.11.0p (build 2011200)
 - Format Self-compiled, changed only platformio.ini - "default_envs = d1_mini"

[image]

twlare, 2020-11-22:
Related to the above, the Presets are not saved in the presets.json file when accessed via the /edit interface.

[image]

twlare, 2020-11-23:
That definitely helped. I haven't been able to reproduce the problem after merging the changes you made today. I will keep testing...

dan-buckley, 2020-11-30:
The presets .cfg has saved a file, but it doesn't look like it 
[image]

If you then press enter after the bracket the json comes into view 
[image]

closed 2021-06-16 by stale[bot]