wled/WLED #1363: Brightness factor is ignored after E1.31 data is finished

twlare, 2020-11-22:
**Describe the bug**
Brightness factor is ignored after an xLights E1.31 effect is finished and control returns to the WLED effect that was running before the xLights data was processed.

**To Reproduce**
In LED Preferences, set Default brightness to 255 and Brightness factor to 10%. Save changes. WLED effects use 10% brightness. Send E1.31 data from xLights. xLights effects use 10% brightness. When the xLights effect ends, WLED will display the previous WLED effect at 100% brightness.

**Expected behavior**
WLED effects should always use the Brightness factor.

**WLED version**
 - Board: Wemos D1
 - Version 0.11.0p
 - Format Self-compiled, changed only platformio.ini - "default_envs = d1_mini"

**Additional context**
It is desirable to use Brightness factor to control LED brightness from programs like xLights. Without a global brightness limiter, each effect in xLights would need to have its brightness adjusted. If you want to raise or lower the brightness of an LED string, instead of changing a single brightness factor, each effect would need to be modified. Other LED controllers such as ESPixelStick have a similar global brightness limiter.

twlare, 2020-11-22:
I also notice that in the LED Preferences page, Brightness factor has an allowable range of 1 to 255 percent. Should it be 1 to 100?

Aircoookie, 2020-11-22:
Thank you, that was an oversight. Fix coming shortly!

1 to 100 will be the most commonly used range, but a factor >100% is supported and can be useful, for example if you sync your lights and run them at half brightness most of the time. If you then have one light that appears dimmer because it has a diffuser, you can run it a bit brighter to compensate. Of course, it is capped at 255 :)

commit ae46eeab (2020-11-22): Fixed invalid save when modifying preset before refresh (might be related to #1361)

Fixed brightness factor ignored on realtime timeout (fixes #1363)
Fixed Phase and Chase effects with LED counts >256 (PR #1366)
closed 2020-11-22 by Aircoookie