wled/WLED #1364: New patterns from Pixelblaze

baerni, 2020-11-22:
Hi. Do you know Pixelblaze? There are some nice effects there, you can download them or see the code. I hope that someone will like some effects and will be added to WLED in the future and available to everyone :) 

Here is the link to the list of effects:
[Patterns](https://electromage.com/patterns/)

Here is a link to a demo video of the effects
[Pattern demo](https://www.youtube.com/watch?v=HF0vpN_qH-A&feature=emb_title)

Here is a link to the project’s website
[Pixelblaze](https://www.bhencke.com/pixelblaze)

benjamw, 2023-09-24:
I'm not sure how feasible the feature would be of allowing users to create patterns as JavaScript code would be, but that feature would be amazing.
Getting the best of both worlds.

blazoncek, 2023-09-24:
Pixelblaze uses proprietary (and closed source) engine for rendering "patterns".
WLED uses different approach to display "effects". Some effects behave as "patterns" but the two are incompatible on general level.