wled/WLED #1369: WS2812B/NodeMCU/WLED only light partial strip

macgiolla, 2020-11-23:
Total newbie with this  - got 3 x ALITOVE 16.4ft WS2812B 5V strips and hooked the 1st strip up to the MCU/power supply. MCU powers up and I am connected to the device via WLED. As for the LED strip, only the 1st 2 feet or so turn on and sometimes a few led's at the far end light up. WLED does update the 2 feet of LEDs just fine (nothing happens the few LEDS at other end).
I've tried all 3 LED strips and they all have the exact same problem. Surely they can't all be bad with the same problem. Tried a different power supply (same brand/model) and still no luck. PSU is a SHNITPWR 5V Power Supply 30A 150W AC/DC Converter Adapter.

Appreciate any help here.
Thanks
Patrick

teedo757, 2020-11-23:
Is the MCU grounded to the same ground as the lights? Also have you gone into LED preferences and change the LED count to reflect how many LEDs you are trying to light?

macgiolla, 2020-11-23:
hi teedo..thx for the response - yes; MCU is grounded to same ground as lights (GND, pin beside Vin and I have data hooked up to D4). I did go into the LED preferences and update the count from 30 to 150 with not obvious change.

Doyle4, 2020-11-23:
Make sure the colors are correct for your leds, some are GRB, RGB, BGR etc, i had to change mine. if wrong way round, they dont always turn on. theres a test in aurdino that lights them up to test which order they are in.

macgiolla, 2020-11-23:
> Make sure the colors are correct for your leds, some are GRB, RGB, BGR etc, i had to change mine. if wrong way round, they dont always turn on. theres a test in aurdino that lights them up to test which order they are in.

Thanks Doyle - the section that does light up is showing the correct colors and i can change the colors via app as needed.

Paulie-Moo, 2020-11-24:
I have the exact same problem, coming along for the ride

Paulie-Moo, 2020-11-24:
I reflashed my board with the led strip connected and that fixed my problem.  May be worth a try

macgiolla, 2020-11-25:
> I reflashed my board with the led strip connected and that fixed my problem. May be worth a try

Thanks for the suggestion. No joy!

shrek1802, 2020-11-25:
What are you using to write to the board?

Do you have another board you can test it with?

Have you checked the led strip does not have a break in any of the lines going to the part that is not lighting up. You could also wire it up from where you are having the fault to make sure it not a faulty Pixel if this is faulty it will not light up after it. If it is a faulty pixel you can cut it out and solder it.

macgiolla, 2020-11-26:
> What are you using to write to the board?
>
> Do you have another board you can test it with?
>
> Have you checked the led strip does not have a break in any of the lines going to the part that is not [...]

tried multiple power supplies and different brand led strips. Same problem.

Using the esphome-flasher -> NodeMCU

macgiolla, 2020-11-26:
Appreciate all the comments/suggestions.

I ordered a different brand of LED's and pretty much had instant success with the new brand. Once I got the new LED strip lit up, I knew my config was all good. So, I went back to the original strip(s) and can not get those to fully function. Must be a bad batch. In any case, those are going back..

closed 2020-11-26 by macgiolla