wled/WLED #1371: Potentiometer and Proximity sensor to light on the W led of SK6812

GACGAMA, 2020-11-24:
Hello everyone!

I've been tinkering with WLED, which seems to be the very best option in controlling aderessable LEDS for ESP modules

I've been using a SK6812 LED strip for 5 years, and my brother helped me create a program with arduino to control these LEDS, but the program got old and WLED do a lot more.

One feature that I'm missing in WLED, and that maybe I will try to recreate, is the use of a potetiometer to light on the W led of RGBW strips as a dimmer. 

This is because when waking up from sleeping, I miss the physical knob to control the lights as they are easier to find when waking up in the night.

Another feature is powering on W led with proximity sensor.

Any ideas on how I could develop this?

Aircoookie, 2020-11-24:
Hi!
Take a look at this usermod to change the brightness with a rotary encoder: https://github.com/Aircoookie/WLED/tree/master/usermods/rotary_encoder_change_brightness

You could probably modify it in order to have it set the White Channel as well, but remember that that is scaled by the set brightness in WLED, so if you only want white light, you'll need to set brightness to max and the RGB color to black beforehand.

A potentiometer would work a bit differently, because you'd need to read in the analog pin and map its voltage to the white value. It would probably be possible to recycle the potentiometer reading part from your old project and use it inside of a WLED usermod :)

korkbaum, 2020-11-28:
Hi Gacgama and Aircookie, 
I also added hardware control to some of my lights following the excellent documentation. Everything works so far but one thing I don't get my head around is how I can make the Web GUI automatically following the hardware setting; i.e. the brightness slider taking on the position according to the setting of the potentiometer. 
A similar question would be to make the GUIs of synced lights follow the controls of the master light. I thought websockets might do the trick but maybe I'm wrong or missing some point. Any hint is much appreciated. 
Best Regards
-korkbaum-

p0ki, 2023-01-29:
Hi, did you manage to make this work? My kitchen under the cabinet led strips just died, well the touch controller died. Will try to fix it first, but playing with the idea to make my own controller with esp32 and a touch sensor (on/off and dimming function) and proximity sensor. I'm a beginner in esp world but i do know electronics... How hard would it be to do something like that

softhack007, 2023-01-30:
* Potentiometer can be used ("analog button")
* Touch sensor is possible on touch-enabled GPIO pins
* PIR proximity sensor could be attached as "button"

Please remember that we have some hardware limitations though:
* ESP8266 on has a single analog pin, so better to use ESP32
* Capacitive touch is only possible on ESP32 on specific pins that have (T)ouch capability.
* Due to hardware limitations on esp32, only GPIO32-39 (ADC#1) can be used as analog input - ADC#2 is in use for WiFi.

Closing this request, as it seems that its possible in WLED 0.14.

closed 2023-01-30 by softhack007