wled/WLED #1381: Filesystem problems with analog MagicHome controller

morphygee, 2020-11-26:
**Describe the bug**
I’m trying to flash a bunch of magic home controllers with wled.

firmware was compiled with esp8285_4CH_MagicHome env and extended with this flags to save memory:
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_ALEXA -D WLED_DISABLE_INFRARED -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE -D WLED_USE_ANALOG_LEDS

flash is successful and the first connect to the wled-AP works.

But:

on the first page I get following warning/error:

SyntaxError:
Unexpected token < in JSON ap position0
and directly after that, on each buttonpress the following message:

Error 10: Could not mount filesystem!
Saving settings and/or presets are also not possible

What works: controlling (on, off, color, 1D effects)

**WLED version**
 - Board: MagicHome 4ch LED controller
 - Version 0.11.0p
 - Format self. compiled binary

**Additional context**
the same error does not ocure with v0.10.2

Aircoookie, 2020-11-27:
Hi!
Sorry about that, the environments now use the correct flash layout for the filesystem :) Give it a try with the latest master!

morphygee, 2020-11-27:
awesome.
works again 
thank you

closed 2020-11-27 by morphygee

luminxa, 2020-12-03:
I'm getting this error without using a magic home controller on the newest release any ideas on what might be causing it or what the fix May be?

Aircoookie, 2020-12-03:
@luminxa what is your ESP model and what binary or environment did you use? This error occurs if the compiled (expected) flash size exceeds the actual flash size.

luminxa, 2020-12-03:
Thanks for the quick response I got it solved. I didn't realize that the new version needed a spiff file system upload before the image :)

luminxa, 2020-12-03:
btw impressed w new version

thedevleon, 2020-12-22:
Just wanted to leave this here for people that run into the same issue:

In your env config, change the `board_build.ldscript` to `common.ldscript_1m128k` instead of `common.ldscript_1m0m`.
The latter config does not have any space for the spiff file system, and as such, it can't be created.

condemil, 2020-12-23:
Probably it makes sense to delete the misleading configuration from here: https://github.com/Aircoookie/WLED/blob/v0.11.1/platformio.ini#L242

Aircoookie, 2020-12-23:
@condemil I agree and will remove the unsupported environments completely.