wled/WLED #1382: ESP2 Bluetooth?

Doyle4, 2020-11-27:
**Is your feature request related to a problem? Please describe.**
To control WLED via Bluetooth using ESP32 :)
A clear and concise description of what you want to happen.
Enable bluetooth, connect to WLED and use

**Describe alternatives you've considered**
NA, Love WLED too much

Thank you for your ideas for making WLED better!

Aircoookie, 2020-12-01:
Hi!
I'm very glad that you like WLED so much :)
This could be done, but it doesn't seem easy: https://esp32.com/viewtopic.php?t=6707
What would be your use case / what makes BLE better than the WLED-AP WiFi access point?  
In any case it would only be useful with some client-side/app code to communicate over Bluetooth as it of course doesn't work in browser (no HTTP)

Doyle4, 2020-12-01:
Thanks for getting back @Aircoookie :)

My friend owns a Gaming Bar and has a lot of consoles etc all ready using wireless and we have found the devices can conflict, tried various channels, so we thought it would good idea to be able to control them via bluetooth instead, I hope someday someone does release a WLED app that works like the web browser.

redakker, 2020-12-06:
> Hi!
> I'm very glad that you like WLED so much :)
> This could be done, but it doesn't seem easy: https://esp32.com/viewtopic.php?t=6707
> What would be your use case / what makes BLE better than the [...]

Hi,

I'm currently working on this kind of solution for my own project.
There is a BluetoothSerial library and a Serial Bluetooth android app for this topic.
You can send commands over Bluetooth to ESP32.
OR
Maybe it could work as a BLE tracker and could change to a personal light effect. If it "feels" some bluetooth device (phone, beacon) around itself could change to greetings color.
But probably this is an overload function for this project. This kind of automation should be done by another device and control WLED board to change the color/effect

JBlackJr, 2021-06-07:
I would also like to see bluetooth protocol incorporated into WLED project! My reasoning: I have installed ~500 LEDs at my family's lake house where they do not have WiFi/ISP. As they are a bit older, they find it troublesome connecting to the AP and and loosing LTE data just to use the lights. As a work-around, I've set up custom IR codes with a remote that has DIY programmable buttons to switch to some of their favorite settings. But would be nice if the APP could communicate via Bluetooth to interface with the controller/backend.

ooglek, 2021-06-28:
I have a very different reason for this. I fly a paramotor and have a 5m strip of LEDs on it. Right now I'm using an SP-105E bluetooth controller for the lights, but want to move to WLED. I use the MagicLED app on IOS. The app sucks, but it is reliable and rock-solid.

Bluetooth is handy because I still want to use data on phone over LTE, and while it may work while I'm flying to have wifi connected to the WLED ESP32, it disrupts the normal flow of things. 

Having the app able to control all the same stuff via Bluetooth instead of WiFi would definitely help in my situation.

When flying I don't want to have to mess with disconnecting and reconnecting networks, power-cycling the ESP32 to get it to reconnect, etc. Bluetooth requires less setup and configuration for a device that will be powered on for 2 hours with a battery and then shut off again, not something that is active all the time.

andregodoy0, 2021-08-08:
I have another scenario for wanting this feature: I have an over desk cabinet that I've added a couple led strips, but I don't have any outlets nearby, so I'm powering it with a power bank and it works great for this purpose (I'm also building my own larger battery in order to provide more amperes and have less frequent recharge cycles).

The problem is that wifi consumes too much energy even with all LEDs turned off (a 10A battery is 50% discharged overnight), so I can't leave it connected all the time and have the automations that I've wanted to (like turn it on automatically when at dawn when I'm working or with the alarm clock in the morning). Adding a relay to the circuit may improve things, but from my research wifi still demands too much energy for a battery operated device.

I've looked into zigbee and BLE controllers/DIY solutions, but WLED is much more feature complete and I'm willing to help with implementation and test of this feature if needed.

djvj1, 2021-09-22:
I'd like to see this as well. I'm trying to use WLED with an ESP32 in my vehicle and found with the chassis fuse in, it's acting as sort of faraday cage and I can't connect to the wifi or it never shows up. As soon as I take the chassis fuse out, all the problems go away. I've tried other bluetooth controllers in my vehicle and they work fine.

blazoncek, 2021-09-22:
Bluetooth will not be implemented (anytime soon) due to huge amount of resources it demands.

ooglek, 2021-09-23:
@blazoncek Would you be willing to roughly outline the work that needs to be done? Maybe I can get the ball rolling with just a piece that moves the functionality closer to completion.

blazoncek, 2021-09-23:
It is the sheer code size and RAM requirements that are prohibiting.

Funkelfetisch, 2022-02-08:
> Hi,
>
> I'm currently working on this kind of solution for my own project. There is a BluetoothSerial library and a Serial Bluetooth android app for this topic. You can send commands over Bluetooth to [...]

did you get any further on this?
we have a BLE app for our Pixelblaze based board (originally on NRF52) and it's esp32 port - but want to move to WLED asap.
This is pretty much a deal breaker, as the wifi stuff is just not practical for endusers.

blazoncek, 2022-02-08:
No. And IMO WiFi is easier than BT. But that is just me.

Apart from my opinion, there was no-one willing to add BT yet. Until someone with interest shows up it may not happen soon. There are caveats as stated above.

snickers2k, 2022-03-07:
+1 to build a few BT connected DIY-Hue-Go or DIY-BT-Obulbs :)

SudoHockey, 2022-05-10:
I would also like to have a Bluetooth solution. Or maybe some work around that doesn't require wifi.  I have a gator that I want to decorate with lights for holidays. Thanks for WLED it is hands down the best!!!!!!! (better than fast led. I use both).

softhack007, 2022-05-21:
I understood that Wifi and Bluetooth cannot be active together on the same ESP32, because they both need to use the same radio/antenna.
 Is this still correct? So, in contrast to the IR receiver solution, you would not have the web UI availeable once that BT (or BLE) is activated? 

That actually means you have two options:

1. Implement _all_ web UI functionality in a new bluetotth API, so that WFi will not be necessary any more (also for configuring WLED, OTA update etc)

2. Implement only a subset of commands for BT, and have some kind of "kill switch" that allows to disable BT and start Wifi to use the web UI

Funkelfetisch, 2022-05-21:
I have no issues with BLE and Wifi on at the same time.
Haven't completed the implementation, but discovery, connect and sending strings back and forth works just fine.
I do have an 8MB ESP32 though - 4 MB won't work!

softhack007, 2022-05-21:
> I have no issues with BLE and Wifi on at the same time. Haven't completed the implementation, but discovery, connect and sending strings back and forth works just fine. I do have an 8MB ESP32 though - 4 MB won't work!

Interesting 👍  ... could you make a usermod for BLE, that works similar to the IR receiver control? Or do you need to have "deeper" control of WLED?

softhack007, 2022-05-21:
Just another idea - in case BLE + Wifi become problematic, or memory requirement are prohibiting - how about using one of these cheap bluetooth boards with serial interface and AT commanding?  Like the HM-10 or HC-05 that we find in all these "arduino robot car" kits?

SteveEisner, 2022-07-17:
Hi all - I'm trying to run with this & my use case is similar to others:  I want to do short BLE communications (not a full WLED UI) between multiple WLED ESP32s, in a place where there's no WiFi AP available.

In a previous non-ESP32 project I used an external nRF24L01 chip but now I'd rather try to leverage the built-in BLE. I've got working BLE mesh code that can run on an ESP32 via NimBLE, but it doesn't work when added to WLED. I've been trying to get the WLED code into "coexistence" mode (where WiFi and BLE can both run on the same antenna by sharing time.)

If anyone has had any luck or wants to collaborate on this, I'm happy to write the code, package it up and contribute it to WLED once it's working. I don't mind hacking WLED core, I'm just hitting issues with the underlying esp stack crashing when I turn on BLE. There's evidence on the 'net that people have gotten wifi+ble coexistence working, but they're typically doing it in esp-idf versus WLED's arduino package.  Maybe we can figure out how to port that over?

PoTheRedTeletubby, 2022-11-21:
I too would like this feature for using WLED in my car. I want to use a bunch of thin strips in my dash that can be used with the music reactivity part of WLED but using a regular microphone would pickup road noise. If I could use bluetooth multi pairing I could play my music to the car speakers and the WLED controller with clean synced audio. I know it might be a huge pain to code this, but if you could that would be amazing. The functionality offered by WLED ESP32 stand alone, or QuinLED-Dig boards is superior to anything else on the market. Its main limitation being tied to WiFi holds it back from going absolutely anywhere, opening tons of new possibilities.

blazoncek, 2022-11-21:
Just use I2S line in from @srg74 if noise is a problem.
BT will not happen any time soon.

softhack007, 2022-11-21:
It seems that two feature requests get mixed up here.

1) control WLED over BT (no internet availeable)
-> as @blazoncek says, this will not happen, as it requires replacing all the web based UIs with a bluetooth command interface. "Everything" includes configuring etc etc,  not just control brightness...
-> in case of no internet, you can still connect to the WLED AP. This works everywhere especially if no internet is availeable, like on a boat or plane.

> If I could use bluetooth multi pairing I could play my music to the car speakers and the WLED controller with clean synced audio

2) this is a different feature request for audioreactive, to have audio input via bluetooth. Unfortunately will also not happen, due to complexity of the coding needed. However there are alternatives
-> with some coding, it should be possible to utilize a second ESP32 as bluetooth to I2S receiver, and wire it to I2S on the WLED esp32. Basic idea is described here: https://github.com/atuline/WLED/wiki/Digital-Microphone-Hookup#use-a-second-esp32-as-bluetooth-audio-to-i2s-device.
-> you can use a bluetooth audio receiver with line-out, and connect that to WLED via line-in to I2S shield. This is already possible today.

I have to admit I don't understand your last comment about quindor's boards. Every ESP32 can run WLED, and ther are many options for ready-made shields of different sizes not just from quindor. Personally i never used any QuinQodWhat dig-dog boards 😉 or similar and still happy with WLED lighting my rooms.

So there are some options already availeable, with a bit of out-of-the-box thinking.

PoTheRedTeletubby, 2022-11-21:
I now see how this is indeed two different types of feature requests. As for audio reactive scenario 2 as it was dubbed. 

I will admit that I am new to this stuff and not particularly familiar with the wiring setup but if you can wire a mic directly into an ESP32 with WLED support then why couldn't you also use a separate Bluetooth 5.0 board with some audio out pins piping the audio into the ESP32 similarly? Using multiple ESP32s just for a singular function add-on is also a bit of an L because you could use that board for another LED setup.

SteveEisner, 2022-11-21:
@ooglek Since posting my comment on Jun 19th, that's actually the path I ended up following.  I was able to get it working, although not exactly what you said, same spirit. My code does send information between WLED instances but it's not exactly hooked into the UDP or AJAX sync endpoints. Instead I used my own smaller protocol to just sync what I needed to sync. It might be possible to connect it to those, though! 

-- Note this is very different than the recent question about getting bluetooth audio! & doesn't solve that. --

I wrote the BLE code first, and got something simple-ish working, but BLE just isn't a super-reliable way to transmit this info since it relies so much on timing windows. There are 3 BLE channels constantly rotating on both sender and receiver, so you have to have a relatively large window where you're scanning and hoping that you'll find an overlap. I messed with this for a bit, trying different windows to see if I could increase the reliability, but moved on to a different approach.

I ended up using ESPNow for broadcast packets instead, and that's been rock solid.  I got 75 individual devices to sync perfectly at ranges up to several hundred feet apart. Then I stopped testing further because I only have 75 ESP boards :) but I don't see any degradation so I bet it could handle many more devices.

There would be some challenges in wiring this up to the actual UDP sync or AJAX endpoints.  Those have payloads that might be larger than ESPNow can handle. Also there's an implicit assumption that one is the "truth" and others are targeted for sync, but that's not the ideal way to run a mesh. Anyway, I'm going to try to package this all up at some point....

blazoncek, 2022-11-22:
Just forget about BT. Look elsewhere. Please.

@SteveEisner we are still waiting for your PR. 😉

PoTheRedTeletubby, 2022-11-22:
@blazoncek I don't quite understand what is so offensive to you about having an open discussion on a tech forum but thanks for being extremely unwelcoming. Others had some interesting points to contribute. There might not be an "ideal" solution to the issue but that doesn't mean it's not worth discussing. This sounds like those business software engineers that always say everything is impossible. Nothing is getting done that way and clearly others have found some work arounds for the other scenario already by not listening to the "forget about BT" line. Obviously my answers aren't here and I could see that without your completely unnecessary comment.

blazoncek, 2022-11-22:
@PoTheRedTeletubby & @rontav I am sorry if my post sounded too blunt.
The fact is that unless someone with experience (that could be you) volunteers to adapt everything for BT it is just not going to happen as there are other areas of WLED that need attention with higher priority.

johne, 2022-12-10:
there are a couple reasons I wanted bt functionality... I want to put lights on my trailer and having wifi setup isn't always practical... I usually don't connect to wifi very often... even at home, i'm not always connected to wifi but still wanna control my lights...

so... I started working on it... I made a user mod that will turn off wifi when set in ble mode (ethernet should still work but I don't have a device yet to test that with)... and has another switch to turn off ble mode... in this, ble mode just exposes all the json api (I have state working so far, but the others should be very easy to add) here is my fork: https://github.com/johne/WLED/tree/ble 

now, please be kind... i'm not usually an embedded developer. I know there are memory leaks (I feel most of them don't matter since I reboot when changing between ble and wifi) and I know there are a ton of formatting problems (anyone have a .prettierrc file for use with c/c++ that they don't mind sharing so I can turn auto-formatting back on)... I also didn't completely understand debugging statements when I first started, so there is a nasty mix of `Serial.println` and `ESP_LOGD` but I do plan to clean those up... even though all that, i'm definitely wanting feedback if you have it... 

my next plan is to take the ios app (I know, apple fan boy, I guess) and make it so the networking layer can switch between ble-json and wifi-json... I haven't started that process yet, but if you are from the future, you can go here: https://github.com/johne/WLED-App/tree/ble to see my progress...

you might say: well, "what about the 600 byte ble limit of Arduino BLE?"  yeah... I'm a hack and just chunked the data into 512 byte chunks... It ain't pretty or fast, but it does seem to work... the cool thing about ble is that I can "push" changes to the phone to keep it up to date with the current state... I might turn that off if I don't see an easy use for it in the app...

softhack007, 2022-12-10:
Hi, yes we can can wait. Just tell us when you'r ready. We're here to help 👍 
Just that we also need to think about long-term maintenance of your usermod, even at times when you might not be interested any more. That's why some coding style (including logging) and lots of comments in the code are really necessary.

About the memory leaks ... please try to fix them if you can. RAM is always low on embedded processors, so any leaked memory is a pity.

> make it so the networking layer can switch between ble-json and wifi-json

Not sure if the phone APP is really using JSON. My undertanding is that the App is basicilly a wrapper around Safari (or another core browser), so a lot of communication will happen with simple HTML POST an GET requests i think.

Cheers, Frank.

johne, 2022-12-10:
yeah... definitely will try to clean up the code when it gets closer... also, the memory leaks (at least the ones I know of) are only stuff initialized at startup and would be deleted at tear down... so I don't think they are costing anything right now cause it is all memory I am using while the app is running... I just mention it cause it has been like 30 years since i've had to even had to deal with new/delete, malloc/free, etc, etc... and i'm a bit... well... rusty

yeah... i'm just discovering now that the app _does_ use the xml api... I may look at making my mod to the app a ble/xml shim... or I may look at making an xml/json shim in the app... or I may look at creating a new app (based on the features of the current one) using react-native (which is what I use for work)...

johne, 2022-12-10:
you are right... the main stuff is just a webview... well... cool in that it stays up to date... bad for my plans... looks like i'm gonna get to make a react-native app... this may take a minute.

Ph-antom22, 2022-12-13:
Can you just make a server for the website so that you can access it on any network?

johne, 2022-12-14:
yeah... there are instructions (https://kno.wled.ge/advanced/remote-access-ifttt/) on how to expose wled out to the world so you don't have to be connected to the same wifi... but that would still not work for a number of the ppl here where they have no connection to the internet... like me and my travel trailer...

johne, 2022-12-29:
ok...I feel like I have it to a pretty good place (minus support for palette details -- but the ui degrades very gracefully).. I found that the ble library takes up alot of program space... so, either I need to give up ota support or move to an 8mb esp32... maybe someone can suggest a way to save space (as previously noted, i'm mostly a "cloud" developer where the sky is the limit for most limits)... I wrote the app in react-native and took the same approach as the other app: used a webview to show the ui... the difference is I inject javascript to switch between "fetch" ing the data with the json api and using ble to get the data... this means my app supports all the ui (except config) of the web app in ble OR wifi mode.... the main list has a switch to switch between ble and wifi... ultimately, i'll want an ethernet version but I can't find a reasonably priced and available esp32 with ethernet and 8mb (maybe someone can suggest something)... I was guessin i'll just solder one up... anyway, my branch/fork is here: https://github.com/johne/WLED/tree/ble for wled and the mobile app is here: https://github.com/johne/WledAppV2 ... right now, the app is only doing ios (which is what I care about), but react-native supports both andriod and ios so it would be easy to get things working for andrioid, i'm guessin....it has definitely been a fun journey... and i'm definitely gonna keep working with wled... my next project is to create a basketball scoreboard (in the spirit of the ping pong clock) to mount behind an acrylic basketball backboard... we love to play in the backyard as a family and it would be nice to have a scoreboard and game timer (maybe even a shot clock if I can work out a reasonable way to reset it during the heat of the game)...

softhack007, 2022-12-29:
> but I can't find a reasonably priced and available esp32 with ethernet and 8mb (maybe someone can suggest something)..

Yes, currently these ESP32 with more than 4MB flash are hard to find. I know two sources on tindie:
* https://www.tindie.com/products/serg74/esp32-wroom-16mb-usb-c-d1-mini32-form-factor-board/
* https://www.tindie.com/products/moonmodules/esp32-wroom32e-16mb-usb-c-d1-mini32-board/

johne, 2022-12-30:
some screenshots from the working app in ble mode... I realized while gathering these that the "info" tab wasn't working, so i'm gonna get on that this weekend.. otherwise, all these features are working ... it also works with wifi devices... as stated above, it doesn't support android yet, but is written in react-native, so should be able to support android with a bit of work... 

[image]

[image]

[image]

[image]

[image]

recbiker, 2023-02-03:
Hi,
   How can i install this on my ios?  Like the option of bluetooth when the LEDS are installed on my bike.

Ph-antom22, 2023-02-03:
Have you started to make this for Android? How much work would it take to make it compatible with android?

Funkelfetisch, 2023-02-03:
is the power consumption lower than on Wifi?
for some projects we have a _very_ limited battery power supply and can't use WLED at all on those due to the >100mA it pulls @5V

johne, 2023-02-06:
> is the power consumption lower than on Wifi? for some projects we have a _very_ limited battery power supply and can't use WLED at all on those due to the >100mA it pulls @5v

I have a battery monitor on my trailer but I also have solar so it is hard to tell during the day... tonight I will switch it back and forth between wifi and ble and see what the power consumption is like with ble vs wifi...

johne, 2023-02-06:
> Hi, How can i install this on my ios? Like the option of bluetooth when the LEDS are installed on my bike. Can this please be published to app store.

you can download the source here: https://github.com/johne/WledAppV2 then look into how to install a react native app to your ios device... unfortunately, you may need an ios developer account to do so... I hope to eventually put this on the app store, but i'm not sure when I would get around to that.

johne, 2023-02-06:
> Have you started to make this for Android? How much work would it take to make it compatible with android?

react native supports ios and android... so there shouldn't be much coding to get it working on android... you can get the code here: https://github.com/johne/WledAppV2 and then see what all steps you need to take to get all the libraries i've used working on android... if you do, please consider contributing that back with a PR...

HackerShurf, 2023-06-11:
I know I am late to the game and it looks like there is some movement on this.  So I have a few applications that would need Bluetooth.  The primary is that I would like to use this for costumes and lighting effects like that.   But I also have some off grid applications that would also benefit from the Bluetooth portion.  I unfortunately do not have the skills yet to help with any coding.

SudoHockey, 2023-06-11:
I have lights on My Tracker (side X Side) That I use while driving around I use 2812b strips with a Dig2Go that has Wled on it. I just connect to it with wifi find the pattern I want and plug it in a usb power stick with Usb c. Unfortunately, I can’t change the pattern but it’s portable. I know this isn’t Bluetooth , but this is what I do. Hope this helps.

TheRealZam, 2023-07-04:
I'm speaking to using BT for lighting control only.  This would be a HUGE game changer for me.  I use multiple wLED instances on my cane / mobility device / cosplay outfits that simply CANNOT plug into one another for mobility issues.  Using WiFi from my phone hobbles my mobile internet, and tethering sucks my phone batteries.  

Couldn't we just run the UI on the app and push the commands via JSON to the BT devices?  Another method would be to run the entire UI in text over BT, but this would cost CPU and be very slow this type of connection.  When in BT mode, multiple wLEDs could also  communicate with one another via either BT or WiFi with one device functioning as a controller.

Ironwrkr, 2023-10-26:
I too would like an android, ble, wled app.  I want to have wled on my boat for Xmas parade.

ooglek, 2023-10-29:
I know this has been a big push by a fair number of users, and I'd still appreciate a Bluetooth Interface.

The ESP-NOW Integration does solve some of the requests for a Bluetooth Interface. 

@TheRealZam when 0.15 comes out, I'm told that you can map an ESP-NOW remote to your WLED interface and use that to change settings. Currently in 0.14 you can only change to 1-4 presets, but I'm told that in 0.15 you can map any action in WLED to any of the remote buttons.

I'm still hopeful that there will be a Bluetooth Interface for WLED someday, as I too would prefer to run an App on my phone that issues commands via Bluetooth.

crosstopher, 2025-06-06:
Sharing mostly as an inspiration at this point, but the following setup is working for me currently as a proof of concept:

- one ESP running a custom build of WLED based on [this fork](https://github.com/insane2subro/WLED-ESPNOW-Json) which processes ESP-NOW messages as WLED JSON API calls.   (with some added functionality to skip the WLED checks for 'authorized' MAC addressess... it'll accept any ESP-NOW messages targeted towards its MAC address without pre-approval..   "promiscuous mode").

- another ESP32 which broadcasts availability over BLE, and is programmed to simply retransmit anything it receives on the BLE channel, verbatim, over ESP-NOW, to any number of pre-specified MAC addresses.  the "translation box".

- a custom iOS app that scans for BLE devices and captures the necessary IDs from selected devices, and then displays a set of buttons that fire off messages containing JSON formatted WLED API commands (power, brightness, presets 1-6, etc), which the translation box forwards to the WLED device.  There's a little complexity due to the fact that the BLE and WiFi share the same hardware on the ESP32 and have to trade off duties as to not interfere with each other.

You just need to have one of these little translation boxes nearby to handle a whole bunch of targets and receivers, but it seems to work pretty well, and requires no WiFi or Bluetooth authentication from any devices, with the assumption that the retransmitter already knows the MAC addresses of any WLED devices it's supposed to contact.

marcodejongh, 2026-11-13:
We want Android and iOS users to control WLED without joining a Wi-Fi network or configuring a VPN.

Would you welcome an optional BLE GATT transport that passes commands to WLED's existing JSON API? Our initial use case is static LED updates, brightness, clearing, and device information.

We'd aim for documented framing, acknowledgements, bounded messages, and configurable access. Wi-Fi coexistence, OTA, and memory use would need hardware validation.

Would you prefer this developed as an external usermod first? We could use Boardsesh as a client to test interoperability.

Context: we've been discussing the phone use case on [WLED #5811](https://github.com/wled/WLED/issues/5811#issuecomment-6849350236). This proposal would use GATT directly rather than requiring the phone to establish an IP tunnel over BLE.