wled/WLED #1386: Adafruit HUZZAH32 Pin 21

boskoe, 2020-11-27:
I read in #1243 that the default data pin in the binaries isn't broken out on the Adafruit HUZZAH32 and that a binary may have been compiled using the HUZZAH32's Pin 21. If a binary was maid I can't seem to find it?

Thanks

srg74, 2020-11-28:
> I read in #1243 that the default data pin in the binaries isn't broken out on the Adafruit HUZZAH32 and that a binary may have been compiled using the HUZZAH32's Pin 21. If a binary was made I can't seem to find it?
> 
> Thanks

Please come over to WLED discord and someone will help you with custom binary.

boskoe, 2020-11-29:
@srg74 Thanks but I managed to borrow an Adafruit Huzzah 8266 which does have the supported pin broken out. I flashed it and it's working very well and I'll use it to learn the WLED enviorment. I have ordered a [Dig-UNO](https://quinled.info/2020/02/11/quinled-dig-uno-pre-assembled-available/) but they are back ordered.

Again thanks for your help.

closed 2020-11-29 by boskoe