wled/WLED #1389: Wifi connection

JudeAbbey, 2020-11-28:
I am using nodemcu flashed with WLED. I am able to connect to the wled-AP and enter my WIFI info - set to 2.4G, 20MHz,802.11b/g/n and  channel 11

After I save my wifi config, I can see the the WLED is assigned an IP  on the router, however I am not able to connect to it.  I tried to ping that IP and there was no response
The wled-AP is no more available.
Any suggestions how to resolve this and get this on my home wifi?

Thank you

Legsmaniac, 2020-11-28:
Did you change the drop-down selection to say always available? 🙂

JudeAbbey, 2020-11-28:
Hmm, No I dont think so, left the default - no connection after reboot.  But disabled wifi sleep. 

Does it need to be set to always to allow it to join home wifi? 
Thanks

Aircoookie, 2020-11-28:
Hi!
If it is assigned an IP but connecting to that is not possible, the signal strength might be the culprit. How far away is the WLED device from your router? You can also try rebooting WLED, sometimes that helps.

JudeAbbey, 2020-11-28:
They are both inside in the same room about 3 meters from each other. Don't  think distance will be an issue. I have done the restart and a couple of reflashing already 

Some more background info:
I have 3 nodemcu with wled. One connected to my wifi but randomly dropped off and now I cant get any to connect.

Aircoookie, 2020-11-28:
Strange... Please use a button on GPIO0 and longpress it for 7 seconds or place the nodemcu outside of the wifi range to make the WLED-AP show up. There you can try if setting a static IP or one of the checkboxes in the experimental section of the WiFi settings helps. What is your router model by the way?

JudeAbbey, 2020-11-28:
Great ! I got the defualt wled AP back by turning my router off. I have set the wled AP to  always!  Saved me from getting on a ladder and also flashing the board again! I am m using hitron coda

JudeAbbey, 2020-11-30:
No luck with the statistic IP. Still cant connect to the wifi.

4ndy8, 2020-12-02:
Could this be related to the UDP issue making network interface unresponsive? [https://github.com/Aircoookie/WLED/issues/1390]

JudeAbbey, 2020-12-06:
Since I am not having luck with getting my mcu on my wifi.. I am wondering  if I can get them on the phone app to work with the wled wifi? Is that possible?

scubastevexj, 2020-12-09:
I have had this exact issue on 11.  I flash 10.2 connect to WIFI, set status and reserve IP on my router, then OTA update to 11.
Just in case it matters, I am using TP link M9+ mesh...

scubastevexj, 2020-12-09:
I actually just tried another nodemcu and I could not get it to connect even going back to 10.2.
troubleshooting included factory reset
reflash 10.2
mess with UDP
turn off 5G on my router leaving only 2.4G
finally i found the issue.

my SSID starts with an "i" and my phone was capitalizing that to "I".
I was still on 10.2, so reflashed to 11 and had no issues connecting once i changed the capitalization.

hl108, 2021-01-03:
Hi, I am facing a similar problem. I am using a Wemos D1 mini running v0.11.1  
My main router (the one connected to the outside) is a Fritzbox 7490. When connecting to its WLAN everything works as expected. However in some rooms I have different APs that are connected to the Fritzbox via LAN cable. When I am in these rooms, the WLED AP comes up initially (as it should), but when I enter the WLAN credentials of the local AP, the device is no longer accessible. The WLED AP does not come up anymore, even after reboot which indicates that the device thinks it is connected. However on the main Fritzbox configuration page I cannot see the device in my home network (it appears in the list of unconnected devices)  and the WLED App cannot find the device.

closed 2021-06-18 by stale[bot]