wled/WLED #1390: UDP Sync causing unsynched devices to drop off network

Morphy99, 2020-11-28:
**Describe the bug**
I have 8 separate WLED strips on my network and I have been looking to sync 3 of them with the UDP sync function. I have the other 5 with Receive Brightness, Color, and Effects unticked and receive UDP realtime unticked as well.
For some bizarre reason this causes the 5 to drop off the network with only a hard reboot resolving it. 

**To Reproduce**
Power on the WLED I want to sync from and press the SYNC button on the main menu via the web gui. Go to Favorites and change preset on this device and the others sync OK. A few seconds later the other 5 devices all drop off the WiFi at the same time. Only way to get them back online is to reboot them manually. The LED's continue to function with effects running if this happens, so I suspect it's a network issue rather than a device crash. 

**Expected behavior**
The other 5 devices to function as normal without synching to the 3 I require to sync.

**WLED version**
 - Board: Wemos D1 x6, NodeMCU v3 x1, ESP-07s x1 (NodeMCU is doing the control of the 2 other synched lights)
 - Version 0.10.2
 - Format Binary APA102

**Additional context**
Things I've tried:

- Changing the "master" sync device in the group of synched devices. Still causes the issue.
- Leaving sync group on for 5 -10 minutes without changing effect or preset doesn't seam to cause the issue.
- Just changing the colour of the master device running colorwaves effect will cause some devices to drop off, strangely one wasn't affected. I've noticed sometimes one of the 5 will stay online but it will eventually go too if I keep changing presets or colours.
Workaround:
- Changing the UDP port on one of the devices out of the sync group still caused it to drop off the first time but I'm thinking it didn't change correctly (maybe needed a reboot) as now it seems to be unaffected by the bug on a different port.

Aircoookie, 2020-12-01:
Thank you for the detailed report!
I suspect that since the UDP interface is always listening, even when turned off, it fills the receive buffer and thus blocks traffic. We can try whether calling `parsePacket()` even when we don't need it will fix this.
I will send you a binary for testing this later today :)

4ndy8, 2020-12-02:
Can confirm that I've just discovered the same issue and not found any alt configurations that mitigate it. Until a firmware update comes, it's sync off for everything or those with UDP listening de-selected will crash. Was surprised that changing the UDP ports didn't provide a work around. Stand alone AP mode I guess is the alternative, but not great with multiple nodes. Great update otherwise tho 👍

Morphy99, 2021-01-02:
> I will send you a binary for testing this later today :)

Did you send this? I thought you had but I've been too busy to test anything and now I can't see any link

closed 2021-06-18 by stale[bot]