wled/WLED #1392: No available sensors on Home Assistant

pepsonEL, 2020-11-28:
Hi
I have installed WLED on my ESP32. Then i added it to Home Assistant integration but i see that:

sensor.wled_uptime

sensor.wled_free_memory

sensor.wled_wi_fi_bssid

sensor.wled_wi_fi_channel

sensor.wled_wi_fi_rssi

sensor.wled_wi_fi_signal

is no available in home assistant and not show state for it. Why ?

sansillusion, 2020-11-29:
You just need to enable them in home assistant. They are disabled by
default.

pepsonEL, 2020-11-29:
Can you tell me how ?

pepsonEL, 2020-11-29:
[image]

sansillusion, 2020-11-30:
Go in configuration->devices click your wled device, show disabled devices,
click on then and flip the enable switch.

closed 2021-06-16 by stale[bot]