wled/WLED #1396: Make it easy to create evenly spaced segments

larrywal, 2020-11-29:
@Aircoookie - Especially with the release of 0.11, I can find myself experimenting with segments a lot!  Frequently I think I'll want them evenly spaced (e.g. 50%|50%, or 25%|25%|25%|25%, etc... maybe 15%|70%|15%).

Would be nice to have a button to auto create the 5-10 most common patterns.  Even if it is only available if resetting segments to seg0 first, I could see this being pretty helpful.

larrywal, 2020-11-29:
Maybe an easy implementation would be a button to create a “fill” segment that starts from 1+ the last used led, and continues to the last. Then it could be modified (or shortened) from there, and then the button could be pressed again.

Aircoookie, 2020-12-01:
This is a really cool idea, thanks! Will add that at some point, it should be quite a simple UI-only change (either a button that allows you to create `n` equally long segments or a button that adds one segment and resizes the other ones accordingly)

larrywal, 2020-12-01:
You bet!  Thanks for the great work on WLED!

blazoncek, 2022-10-10:
Implemented in 0.14

closed 2022-10-10 by blazoncek