wled/WLED #1397: Lights wont turn on automatically, and ramdomly turns off when it does

seanymack, 2020-11-30:
I have 2 ESP8266 WLED controllers running christmas lights and i set a simple time macro to turn on at 17:00 (T=1) and turn off at midnight (T=0) but its not working reliably. Sometime they work, and sometimes they dont. Also, when i turn on the lights manully, after an hour or so they turn themselves off. Could it be some simple setting i am missing? Timer was not active 

Both controllers are in sync and preset cycle is turned on 

Thanks for your help

fribse, 2020-11-30:
Not sure about the failing to turn off, sounds like a wifi problem, but the second one with them turning off after an hour, that sounds like this
[image]
under LED preferences?

seanymack, 2020-12-01:
i checked and timed option was not turned on. Does the controller need to have a strong wifi for it to turn on/off? i didnt think so

Last night i removed the time macros, to my surprise the lights stayed on all night and into the morning, which suggests its some setting in the time and macro's section that's causing the lights to turn off after an hour or so

This is what my settings look like

[image]
[image]

Aircoookie, 2020-12-01:
Those settings seem alright. I don't know what it is which is turning them off unexpectedly 🤔

closed 2021-06-16 by stale[bot]