wled/WLED #1398: WLED 0.11, multi candle effect fails to set second colour in preset

fribse, 2020-11-30:
**Describe the bug**
I have a "candle multi" effect I put on the garden lights, at very low level during the night.
I've tried saving it as a preset. But when doing so, it only saves the primary colour, which then makes the "candle multi" not do anything. It also seems that if I set a secondary myself afterwards, it resets the effect to solid. So I have to set the two colours and then set the 'candle multi', not sure if that is why it can't set it properly from the preset.

**To Reproduce**
Set this:
```
<vs>
<ac>128</ac>
<cl>255</cl>
<cl>160</cl>
<cl>0</cl>
<cs>255</cs>
<cs>237</cs>
<cs>38</cs>
<ns>1</ns>
<nr>1</nr>
<nl>0</nl>
<nf>1</nf>
<nd>60</nd>
<nt>0</nt>
<fx>102</fx>
<sx>128</sx>
<ix>128</ix>
<fp>0</fp>
<wv>0</wv>
<ws>0</ws>
<ps>0</ps>
<cy>0</cy>
<ds>Garden effectlight</ds>
<ss>0</ss>
</vs>
```
Save it as a preset, change to another effect, reload the preset.

**Expected behavior**
It's just supposed to save the colours, and recall them, and the effect, speed, intensity and brightness.

**WLED version**
 - Board: esp32 (d1 mini format)
 - Version: 0.11.0
 - Format: Binary

Aircoookie, 2020-11-30:
Hi!
The XML response to the `/win` command is not a valid API command.

Try saving `A=128&R=255&G=160&B=0&R2=255&G2=237&B2=38&FX=102&SX=128&IX=128` (or easier, saving the preset in the UI with `Use current state` checken when the Candle Multi effect is applied) and it should work :)

fribse, 2020-11-30:
Hi @Aircoookie 
Well, you should blame your own code then for the incorrect response to the HTTP API :-D
I've tried setting the effect in the GUI, and then save it as a preset, and THEN document it with a /win call ;-)

Aircoookie, 2020-11-30:
Ah okay :) Sorry for the confusion, but it is documented that the HTTP API responds differently to what you would need to send to change values.
Please send me your `[WLED-Ip]/presets.json` file, so I can take a look :)

fribse, 2020-12-01:
Ok, I think I got it set after doing several sets, this is the json file:

`{"0":{}                                                                                                                                                                                                                                                                                                                                              ,"15":{"n":"Preset 15","bri":128,"seg":{"col":[[8,37,255],[255,18,65],[49,255,3]],"fx":80,"sx":128,"ix":224,"pal":0}},"1":{"n":"FCK","on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":657,"grp":1,"spc":0,"on":true,"bri":255,"col":[[245,255,247],[4,0,255],[0,0,0]],"fx":37,"sx":8,"ix":255,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]},"2":{"n":"Candle Multi","on":true,"bri":20,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":657,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,160,0],[255,255,36],[0,0,0]],"fx":102,"sx":128,"ix":209,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}            ,"3":{"n":"Merry Christmas","on":true,"bri":62,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":657,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,160,0],[255,245,54],[237,255,237]],"fx":44,"sx":3,"ix":255,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}}
`

Aircoookie, 2020-12-01:
Yep, your `Candle Multi` preset should do the correct thing!

yetdog, 2020-12-03:
I actually don't see a change between Candle and Candle Multi with 0.11. Both effects seem to behave like just plain old "Candle" - the multi behavior seems to be gone in 0.11, even though the effect is still listed as an option.

Aircoookie, 2020-12-03:
@yetdog Candle Multi should still work as intended. However, it uses a lot of memory, so there is a cap in WLED so it doesn't use too much. If you have over ~700 LEDs in the segment, or are low on memory, Multi Candle willl fall back to the standard Candle effect.

yetdog, 2020-12-03:
> @yetdog Candle Multi should still work as intended. However, it uses a lot of memory, so there is a cap in WLED so it doesn't use too much. If you have over ~700 LEDs in the segment, or are low on memory, Multi Candle willl fall back to the standard Candle effect.

That's absolutely my issue; I'm currently running 6 x 5m/150LED 5v strips - so 900 LEDs run by a single ESP8266. Thanks for helping my understanding!

Aircoookie, 2020-12-03:
Either use an ESP32, or you can set the Segment Grouping to 2! That way, every two adjacent LEDs will flicker in unison and the effect should work correctly. Depending on your setup, an even larger grouping may also look great!

fribse, 2020-12-04:
I have oddly enough 587 in my christmas sign (with an esp32) and 200 with my lightchains, so accidently I avoided those limitations :-)
[image]

closed 2021-06-04 by fribse