wled/WLED #1399: Choice of PINS

Legsmaniac, 2020-11-30:
First of all, well done on a great new major release and just in time for Xmas!  👍
Love the new Presets, haven't had a full and proper play with everything yet but what I've checked over - brilliant.
Well done Christian and the team and thank you.

It's great that you've also provided a small selection of ready compiled files with a choice of PINS to use.
However, I was under the impression that it was going to be a selectable option in the GUI Config page?
Seems I am wrong but slightly disappointed it wasn't included.  It would be great to be able to change the PIN number on-the-fly without having to recompile and flash or upload each time.  I know some of the less savvy users wanted this.  Not so much a problem for me as I can recompile the code but it certainly would save a lot of time.
If I knew a bit more about coding or felt confident enough to tackle it I'd have a go myself.  It's gotta be easy enough to add in I would have thought?

Anyway, it's only a minor thing, to me at least.

All the best, stay safe and have a great Christmas!
😊

Aircoookie, 2020-11-30:
Thanl you :)
This is in fact the mainline feature coming in v0.12.0. It isn't trivial since different pins need different drivers, but it's coming, alongside with dynamically choosing which type of strip (WS2812, APA102, WS2801, ...) you have and even the support for multiple strips on one controller!

You too 😊

Legsmaniac, 2020-11-30:
Ah!  V12.  OK, Excellent.  Thank you.  :-)

closed 2020-11-30 by Legsmaniac