wled/WLED #1400: Slider Annoyance on Phone

Doyle4, 2020-11-30:
**Describe the bug**
Using color slider changes webpage

**To Reproduce**
Slide the color slider

**Expected behavior**
To choose colors with slider without it going it going to home page or effects page if sliding to the right.

**WLED version**
 - Board: Doesn't matter
 - Version 11
 - Format Binary

**Additional context**
Possible to have the bottom row, color, effects etc to not accept "swipes" out of their command zone (don't know how else to describe it sorry), when not hovering over them, any swipes on the screen do not interfere with them.

Thank you for your help!

Legsmaniac, 2020-11-30:
Hmmmmm.  I cannot reproduce this.

Aircoookie, 2020-11-30:
Thank you for the report!
Could not reproduce it either yet, does this happen with the sliders underneath the color wheel or with the optional RGB sliders? What is your phone model/browser/app?

Doyle4, 2020-11-30:
Trying to upload a video, will try when home.
Googlechrome on iPhone :)

Doyle4, 2020-11-30:
I think I know what it is, my finger is touching the outside of the colour slider box and moving across to different section

huggy-d1, 2020-11-30:
Try using one of those touch sensitive pen tops (often black & foam-like).

allgoewer, 2020-12-07:
I notice the same issue on a Pixel 4a running Android 11, in the WLED-App (1.03) as well as in Chrome (87.0.4280.86)
See https://www.youtube.com/watch?v=vkNUljU3Rk4

It seems not to happen with the white-channel slider

Doyle4, 2020-12-07:
So glad someone else has spotted it! It keeps happening and I thought it was me.

allgoewer, 2020-12-08:
I think I found the issue.

This patch https://github.com/Aircoookie/WLED/blob/ac010cd7b21cba8e2d91d01ca6a100050879e248/wled00/data/index.htm#L1128 seems to update iro.js

If you blame this line with a previous version, you'll notice that the [lock()](https://github.com/Aircoookie/WLED/blame/362156c8abb6f4580ff2edff8b49a651e946c0ab/wled00/data/index.htm#L2025) function checks for some classes provided by iro.js, but it seems as these names changed.

I'll try to fix it and create a pull request later.

Doyle4, 2020-12-08:
Keeping an eye on this as its driving me mad, few customers with devices I have made are complaining of it also. Told them a patch is coming soon.

JCPTECH, 2020-12-09:
I am also getting this issue. for colour wheel and sliders. How to patch this issue?

allgoewer, 2020-12-09:
Well, you could

* Go back to v0.10.x
* Wait for my PR to be merged into (hopefully) the next patch release
* Build an image with my patch yourself

closed 2020-12-09 by Aircoookie

Doyle4, 2020-12-09:
Thanks for looking into this @allgoewer and thanks @Aircoookie also :) 

Great team and support.

PR #1459 (2020-12-09): Fix #1400: Using sliders / colorwheel sometimes slides the whole UI
Hi @Aircoookie,

this is an attempt to fix #1400 .

iro.js has been updated in an earlier commit which changed the names of the iro-classes.
This led to the function check() in wled00/data/index.htm no longer correctly detecting whether the touch event originated from one of the sliders or the colorwheel.

My initial attempt to simply update the class names did not work as expected.
While this reduced the issue, it did not completely remove it.

Turns out some of iro.js's elements (such as the ring that makes up the slider / c [... cut]
commit 7684fb83 (2020-12-09): Fix #1400: Using sliders / colorwheel sometimes slides the whole UI (#1459)

* Prevent UI from sliding when using sliders (#1400)

An iro.js version bump in commit 98e4ac6 changed names for the slider &
colorwheel classes which made the filter in function lock() fail.

Also it seems that not all elements of the Iro sliders & colorwheel have
appropriate classes set. Just changing the names still sometimes failed
when the user accidentally grabbed a part of the slider / colorwheel
that has no class name set.

To mitigate this, lock() now checks the events classlist for classes
starting with "Iro [... cut]