wled/WLED #1401: Alexa doesn't find a single WLED device in my house.

salopette, 2020-11-30:
Hello, I have a little problem, my Echo Alexa cannot find a single Wled device in the house, I have 6 different ones!
Everywhere the last version of Wled is flashed as a bin file.

Do you still have to change / adapt something?

I have activated Alexa in the Wled settings!

[image]

huggy-d1, 2020-11-30:
What type of alexa enabled device(s) are you trying to use?  Echo Dot, Flex, Show, and so on?  If you have an Echo Dot, power it off, wait a few seconds, and power it back on again.  Then ask it to discover new devices.

salopette, 2020-11-30:
I have echo dot, echo, echo show, FireCube, Echo Flex. But nothing is found.

huggy-d1, 2020-11-30:
Are you able to connect to the WLED instances from PC, smartphone, tablet without issue?  Are you able to use the alexa app on your smartphone / tablet without issue?  
Is your network configured with IoT security to keep WLED away from the rest of the network?

huggy-d1, 2020-11-30:
I think you need to disable UDP receive on any device you want Alexa controllable.  For now, disable UDP sync on all your WLED installs so you can try and discover them.

salopette, 2020-11-30:
Yes, I can access the Wled devices via the IP without any problems, the app on Android also works without problems.
my network is fritzbox 7590 without restrictions.

huggy-d1, 2020-11-30:
Maybe this link will help: https://wled.discourse.group/t/wled-going-offline/1289/12

salopette, 2020-11-30:
-I tried, it doesn't help.

[image]

huggy-d1, 2020-11-30:
This link help? : https://github.com/Aircoookie/WLED/issues/1283
Here's the extent of my searching WLED github issues for Alexa:
1. hostname (the one the mDNS of the router and your phone might use) needs to be basically alpha-numeric, no spaces, no special characters;
2. same for alexa name
3. turn off WiFi sleep mode
4. if toggling power on a 3rd generation DOT does not enable discovery of a single WLED install, open the admin of the router and remove all learned WLED names / IPs (while they are powered off).  Power off the DOT.  Turn on a single WLED instance.  Verify you can connect to it via smartphone, then disconnect from it.  Now power up the DOT and discover new devices as soon as it will allow.
Repeat for each WLED device as needed.

If you find a definite approach that works, please do post back your results.

It seems to mostly rely on naming simply, making sure it does not go to sleep, making sure WLED and DOT are on the same network (not just pingable), and 2.4 and 5GHz are not joined together into a single SSID.  Some others mentioned they needed to enable "legacy mode" on UniFi hardware.

Aircoookie, 2020-12-01:
Alexa support unfortunately is a bit broken right now :(
Sometimes you might be able to get it to discover if you reset the WLED light, reset the Echo (Dot 2), and immediately tell it "Suche Geräte" once it stops the blue spinny boot animation. Perhaps it will work, but you are by far not the only one where it doesn't work correctly. I hope I have time to troubleshoot this at some point :)

derekpurdy, 2020-12-01:
> It seems to mostly rely on naming simply, making sure it does not go to sleep, making sure WLED and DOT are on the same network (not just pingable), and 2.4 and 5GHz are not joined together into a single SSID. Some others mentioned they needed to enable "legacy mode" on UniFi hardware.

In the latest changes with Unifi and how they are updating their UI, it appears you cannot delete upnp devices nor can you enable legacy mode. Atleast I cannot find it.

WolJG, 2020-12-09:
I think this is part of this issue: I have 2 "old" WLED-devices that work with Alexa over a year. Now i want to add a 3rd but Alexa doesn´t find it. I unplug an echo dot 3rd and plug it in again, then Alexa finds the 3rd device. But now when i handle 3rd device via Alexa the "old" 2 devices react in the some manner (on/off/color/brightness). So the 2 old devices are synced to the 3rd via Alexa.  - strange -
When i handle the old devices everything is fine (no sync) 
Tested with 10.2 and 11.0 although SR-fork. Always the same. Seems to be a native Alexa-problem. What do you think?

closed 2021-06-16 by stale[bot]