wled/WLED #1402: ESP32 multistrip usermod failing to compile on v0.11.0

andyshinn, 2020-11-30:
**Describe the bug**
It looks like https://github.com/Aircoookie/WLED/commit/fb91d9b2d483c978c986a6f9999b5cf797e3a12c broke the esp32_multistrip usermod. Compiling the usermod now fails on 0.11.0.

**To Reproduce**
1. `git checkout v0.11.0`
2. Create custom `platformio_override.ini`:
```
[env:quinled_dig_quad]
extends = env:esp32dev
build_flags = ${common.build_flags_esp32}
  -D USERMOD_DALLASTEMPERATURE
  -D USERMOD_DALLASTEMPERATURE_CELSIUS
  -D ESP32_MULTISTRIP
  -D NUM_STRIPS=2
  -D PIXEL_COUNTS="200, 108"
  -D DATA_PINS="16, 1"
```
3. `pio run -e quinled_dig_quad`

**Expected behavior**
Compiling firmware.

**WLED version**
 - Board: Generic Wemos D32 mini clone
 - Version: v0.11.0 tag
 - Format: Self-compiled

**Additional context**
Error during build:

<details>
```
wled on  master [?] is 📦 v0.11.0 via ⬢ v12.19.0 via 🐍 v3.7.9 (python-venv-3.7.9) 
❯ pio run -e quinled_dig_quad
Processing quinled_dig_quad (board: esp32dev; platform: espressif32@2.0; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting wled00.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <AsyncTCP> 1.0.3
|-- <IRremoteESP8266> 2.7.3
|-- <LittleFS_esp32> 1.0.5+sha.b941415
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 2.0.0+sha.3336275
|   |-- <AsyncTCP> 1.0.3
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DallasTemperature> 3.9.1
|   |-- <OneWire> 2.3.5
|-- <OneWire> 2.3.5
|-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Building in release mode
*** use existing my_config.h ***
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/FastLED.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/colorpalettes.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/bitswap.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/colorutils.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/hsv2rgb.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/lib8tion.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/noise.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/platforms.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/power_mgt.cpp.o' from cache
Retrieved `.pio/build/quinled_dig_quad/lib886/FastLED/wiring.cpp.o' from cache
Compiling .pio/build/quinled_dig_quad/src/FX.cpp.o
Compiling .pio/build/quinled_dig_quad/src/FX_fcn.cpp.o
Compiling .pio/build/quinled_dig_quad/src/alexa.cpp.o
Compiling .pio/build/quinled_dig_quad/src/blynk.cpp.o
Compiling .pio/build/quinled_dig_quad/src/button.cpp.o
Compiling .pio/build/quinled_dig_quad/src/cfg.cpp.o
Compiling .pio/build/quinled_dig_quad/src/colors.cpp.o
Compiling .pio/build/quinled_dig_quad/src/dmx.cpp.o
wled00/FX_fcn.cpp: In member function 'void WS2812FX::show()':
wled00/FX_fcn.cpp:251:26: error: 'class NeoPixelWrapper' has no member named 'GetPixelColorRaw'
       RgbwColor c = bus->GetPixelColorRaw(i);
                          ^
wled00/FX_fcn.cpp: In member function 'uint32_t WS2812FX::getPixelColor(uint16_t)':
wled00/FX_fcn.cpp:460:34: error: cannot convert 'RgbwColor' to 'uint32_t {aka unsigned int}' in return
   return bus->GetPixelColorRgbw(i);
                                  ^
wled00/FX_fcn.cpp: In member function 'uint8_t WS2812FX::getColorOrder()':
wled00/FX_fcn.cpp:481:15: error: 'class NeoPixelWrapper' has no member named 'GetColorOrder'
   return bus->GetColorOrder();
               ^
wled00/FX_fcn.cpp: In member function 'void WS2812FX::setColorOrder(uint8_t)':
wled00/FX_fcn.cpp:485:8: error: 'class NeoPixelWrapper' has no member named 'SetColorOrder'
   bus->SetColorOrder(co);
        ^
*** [.pio/build/quinled_dig_quad/src/FX_fcn.cpp.o] Error 1
============================================================= [FAILED] Took 5.00 seconds =============================================================

Environment       Status    Duration
----------------  --------  ------------
quinled_dig_quad  FAILED    00:00:05.001
======================================================= 1 failed, 0 succeeded in 00:00:05.001 =======================================================
```
</details>

Thank you for your help!

huggy-d1, 2020-11-30:
Something is strange here.  In your post, to reproduce it says to do steps 1/2/3.  However, in the detailed compiler output, it shows compiling cached library data.
First, get steps 1/2/3 working.  That means Step 0: move existing libraries / files related to QuinLEDdigQuad somewhere else.
Then integrate the QuinLED Dig Quad source into the compilation process.

andyshinn, 2020-11-30:
Is there a cache other than `~/.platformio` and `.pio` that I am missing? I was under the impression that those are not really cached but come from PlatformIO installing them at runtime. I've updated the output to show them being installed. The error is still the same.

andyshinn, 2020-11-30:
Ah, I missed the original PR already had a mention of it: https://github.com/Aircoookie/WLED/pull/1362. I'll test out the updated branch to provide feedback.

andyshinn, 2020-12-01:
Fixed in #1409!

closed 2020-12-01 by andyshinn

poldim, 2020-12-10:
Just adding a note here incase anyone else runs into this.  I ran into the below error after following your steps.  I took a guess and created the `src` folder and tried to compile and it all worked.

```
Building in release mode
*** use existing my_config.h ***
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/FastLED.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/bitswap.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/colorpalettes.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/colorutils.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/hsv2rgb.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/lib8tion.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/noise.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/platforms.cpp.o' from cache
Retrieved `.pio/build/esp32_multistrip/lib15b/FastLED/power_mgt.cpp.o' from cache
Compiling .pio/build/esp32_multistrip/lib15b/FastLED/wiring.cpp.o
Compiling .pio/build/esp32_multistrip/src/FX.cpp.o
Compiling .pio/build/esp32_multistrip/src/FX_fcn.cpp.o
Compiling .pio/build/esp32_multistrip/src/alexa.cpp.o
Compiling .pio/build/esp32_multistrip/src/blynk.cpp.o
Compiling .pio/build/esp32_multistrip/src/button.cpp.o
Compiling .pio/build/esp32_multistrip/src/cfg.cpp.o
Compiling .pio/build/esp32_multistrip/src/colors.cpp.o
Compiling .pio/build/esp32_multistrip/src/dmx.cpp.o
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/FX_fcn.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/FX_fcn.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/button.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/button.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/colors.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/colors.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/blynk.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/blynk.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/dmx.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/dmx.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/alexa.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/alexa.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/FX.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/FX.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_multistrip/src/cfg.cpp.o: No such file or directory
*** [.pio/build/esp32_multistrip/src/cfg.cpp.o] Error 1
FileNotFoundError: [Errno 2] No such file or directory: '/Users/poldim/Documents/Other/Projects/WLED/.pio/build/esp32_multistrip/.sconsign38.dblite':
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/Main.py", line 1391:
    _exec_main(parser, values)
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/Main.py", line 1355:
    _main(parser)
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/Main.py", line 1126:
    nodes = _build_targets(fs, options, targets, target_top)
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/Main.py", line 1329:
    jobs.run(postfunc = jobs_postfunc)
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Job.py", line 113:
    postfunc()
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/Main.py", line 1326:
    SCons.SConsign.write()
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/SConsign.py", line 114:
    syncmethod()
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 144:
    self._os_unlink(self._file_name)
Exception ignored in: <function dblite.__del__ at 0x11085a550>
Traceback (most recent call last):
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 126, in __del__
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 123, in close
  File "/Users/poldim/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 144, in sync
FileNotFoundError: [Errno 2] No such file or directory: '/Users/poldim/Documents/Other/Projects/WLED/.pio/build/esp32_multistrip/.sconsign38.dblite'
================================================================================= [FAILED] Took 13.52 seconds =================================================================================

Environment       Status    Duration
----------------  --------  ------------
esp32_multistrip  FAILED    00:00:13.523
============================================================================ 1 failed, 0 succeeded in 00:00:13.523 ============================================================================
The terminal process "pio 'run'" terminated with exit code: 1.
```

Aircoookie, 2020-12-10:
@poldim thank you! that is a known issue, see: https://github.com/Aircoookie/WLED/wiki/Compiling-WLED#compilation-guide-platformio
Just compiling again without changing anything makes it work. Not been able to figure out the cause of it yet...