wled/WLED #1403: OTA update improvement

andywarburton, 2020-11-30:
So found a weird little bug today. I wanted to upgrade my ESP8266 powered WLED strips using the OTA upgraded to the latest version. I initially opened the app and tried to do it there (running on an iPhone 11 Pro with the bin file stored in my “files”). Within the app I got this error:

[image]

“Device unreachable”

But if I went directly to the lightname.local I could do the upgrade just fine

In retrospect, I don’t think it was failing but because the webview in the WLED app blocks the browser chrome you don’t see the upload progress and it looks like it has just hung as you get no indication something is happening.

So... if that’s the case, when someone presses the “update” button it could be worth disabling the “update” button and showing a faux progress indicator or spinner so impatient little monkeys like me don’t keep slapping the update button!

Aircoookie, 2020-12-01:
Thank you! Yeah, I believe replacing the buttons with a simple `Uploading...` text would make sense. Spamming the button could even interfere with the update working correctly, so this is definitely a good idea!

f1vefour, 2020-12-06:
I agree, I updated to the awesome 0.11 and had to fight the urge to spam the button as there was no feedback which made me think I missed the button.

blazoncek, 2022-07-25:
Redirection implemented after button press.

closed 2022-07-25 by blazoncek