wled/WLED #1404: Preset delete confirmation

ToXIc-Dev, 2020-11-30:
**Is your feature request related to a problem? Please describe.**
Accidental deletion of presets

**Describe the solution you'd like**
A confirmation prompt

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thank you for your ideas for making WLED better!

Aircoookie, 2020-12-01:
Thank you for the suggestion!
I thought about adding a confirmation as well, but did not because I think it would make mass-deleting (cleanup) of old presets a lot more annoying.
In day-to-day use, you also still need to click the little down-facing arrow to expand the preset options before you can delete it, so accidental deletion could only happen if you are currently editing a preset.
I could add an optional confirmation though if accidental deletion turns out to indeed an issue.

ToXIc-Dev, 2020-12-01:
@Aircoookie You make some good points, maybe it's not that big of an issue.
I just thought it could be a possible issue but now I think about it, it's probably fine.

huggy-d1, 2020-12-01:
How about an "undo" delete option - only available as the next command after a deletion.  After anything else, the Undo Delete option goes away.

Aircoookie, 2020-12-01:
@huggy-d1 good idea! We can simply temporarily save the deleted preset and restore it with a single click

blazoncek, 2023-01-25:
Preset delete confirmation is implemented. For "undo" please open new issue.

closed 2023-01-25 by blazoncek