wled/WLED #1405: Edit Segment Names?

3ative, 2020-11-30:
**Is your feature request related to a problem? Please describe.**
None, everything is great. 👍

**Describe the solution you'd like**
Now that we have **Pre-set** names  - _GREAT FEATURE BTW_ 👌
Would it be possible to be able to name the **Segments**, in the same way?

Thank you for your ideas for making WLED better!

Aircoookie, 2020-12-01:
It is a great idea I have thought of while implementing preset names actually :)

Will add at some point, but I didn't really want the segment names to be included in every single preset, so I need to get creative on that!

3ative, 2020-12-01:
@Aircoookie 
I didn't think of that. Of course, that'll cause the pre-set settings size to increase by quite a bit. 

Hmm... Could the names be saved in the 'Segments' themselves - There's already 'Segment x' and even 3-4 extra characters would be helpful and I can't think of a need to have different segment names for each pre-set.

sansillusion, 2020-12-01:
Creating a list/table to match id of segment when you meed to output the
name. That way minimal code change is needed, just use the table to match
segment id with a name.

3ative, 2020-12-01:
@sansillusion 
I'm not sure I follow but, isn't that more coding than just making the segment name a variable?

Aircoookie, 2020-12-01:
@3ative of course it is more coding (the reason this feature doesn't exist yet), but it uses less resources (both in RAM and preset file) once done :)

3ative, 2020-12-01:
@Aircoookie Cool Cool
I differ to your expert knowledge - _I couldn't even get a bin file compiled_ **LMAO**

rmp63526, 2021-06-20:
Any news yet?
is there any solution?

blazoncek, 2022-07-25:
Implemented in #2184

closed 2022-07-25 by blazoncek