wled/WLED #1406: Next preset, LP=~, restarts after preset 5

FineLumps, 2020-11-30:
**Describe the bug**
Before 0.11.0, I used the API commands as macros in the Time-controlled Presets section to turn the lights on and off and select the next preset daily.  I've attempted to recreate that functionality in 0.11.0 with presets 50 and 51.  These are exact command copies of my previous macros.

My presets:
ID 1 - preset I want to use regularly but manually set, power on default

ID 3 - preset I want to rotate through
ID 4 - preset I want to rotate through
ID 5 - preset I want to rotate through
ID 6 - preset I want to rotate through
ID 7 - preset I want to rotate through

ID 50 - API command ND
ID 51 - API command T=1&LP=~
ID 52 - API command LP=~

ID 100 - preset I want to use for special occasion and manually set.

When selecting/clicking preset 51 (or 52) the preset selection goes like this:
ID 1
Error 12 (understandable as there is no preset 2)
ID 3
ID 4
ID 5
ID 1
Error 12
ID 3
ID 4
ID 5
ID 1
...

**To Reproduce**
Not sure

**Expected behavior**
Minimally, I would expect LP=~ to cycle through all defined presets.
Ideally, I would like to use something like LP=~ to cycle a predefined range, in my case 3-7, and, along with time-controlled presets, have a new preset selected daily.

**WLED version**
 - Board: ESP8266
 - Version: 0.11.0
 - Format: Binary

**Additional context**
I worry about the behavior of cycling to a preset with API commands like ND.  I don't want my lights to turn on with T=1&LP=~ only to select the next preset with the command ND and have the lights immediately start the turn off timer.

Thank you very much for your help!  Loving WLED.

Aircoookie, 2020-12-01:
Hi!
Please try the API command `PL=~&P1=3&P2=7` for both cycle presets (50 and 51, in 50 of course with the T=1), that should set the correct range for the preset cycle :)

FineLumps, 2020-12-01:
Thank you so much for replying.  I see now that the Preset cycle options for first and last preset are enabled even without the Preset cycle checkbox checked.  Your solution worked.

closed 2021-06-16 by stale[bot]