wled/WLED #1407: Cannot compile any of analog env

srg74, 2020-11-30:
**Describe the bug**
When I'm trying to compile any of env for analog strip error occur
<code>wled00/FX_fcn.cpp: In member function 'void WS2812FX::setRgbwPwm()':
wled00/FX_fcn.cpp:931:7: error: 'color' was not declared in this scope
   if (color == _analogLastColor && b == _analogLastBri) return;
       ^
wled00/FX_fcn.cpp:951:22: error: 'color' was not declared in this scope
   _analogLastColor = color;</code>

**To Reproduce**
Try to compile any analog env with USE_ANALOG_LEDS

**Expected behavior**
Firmware have to compiled

**WLED version**
 - Board: Any analog
 - Format self compiled latest version

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

huggy-d1, 2020-11-30:
Related: https://github.com/Aircoookie/WLED/pull/1362 ??

srg74, 2020-11-30:
Fixed

closed 2020-11-30 by srg74